Atmosphere-NX/Atmosphere #1225: 0.16.0 Pre-Release: Issue with emuMMC

Remna0w0, 2020-12-02:
### What's the issue you encountered?

When running Atmosphère with emuMMC enabled, Atmosphère crashes on boot after the Nintendo Switch Logo appears. This only occurs with emuMMC enabled, as I've tested with it disabled and enabled several times, and it has only crashed when emuMMC was enabled. 

### How can the issue be reproduced?

While emuMMC is enabled, (I enabled/disabled mine through Hekate), run the latest fusee-primary.

### Crash Report

https://gist.github.com/Remna0w0/b9b365f3cf3afe548adb6852ff029b0b 

### System Firmware Version

11.0.0</br>

### Environment?

- Launched fusee-primary payload through Hekate
- Version 0.16.0 pre-release.

### Additional context?

N/A

Skonikol, 2020-12-02:
Do you have a custom theme installed?

Remna0w0, 2020-12-02:
> Do you have a custom theme installed?

I do actually. The custom theme worked fine when I ran it on sysMMC, but my sysMMC isn't updated to 11.0,0. My emuMMC is. Should I uninstall it?

flooki000, 2020-12-02:
emuMMC is disabled but it never work always fatal error !!

Remna0w0, 2020-12-02:
> > Do you have a custom theme installed?
> 
> I do actually. The custom theme worked fine when I ran it on sysMMC, but my sysMMC isn't updated to 11.0,0. My emuMMC is. Should I uninstall it?

Ok so I uninstalled the theme and it works now! Must be because the Home screen has been changed, thanks!

closed 2020-12-03 by SciresM