Atmosphere-NX/Atmosphere #1227: Fatal Error

Gilgn, 2020-12-03:
I'm having trouble trying to use atmosphere. Every time I run it I keep getting this error "[ERROR] Fatal error: [NXBOOT] Couldn't parse boot0: Illegal byte sequence!   Press POWER to reboot" and every time I do press the power button I keep getting this same message. I don't know what to do, did I install something wrong or am I missing something. Please Help!

SciresM, 2020-12-03:
You're using hekate, which hasn't been updated for 11.0.0: use fusee primary or wait for a hekate update.

SciresM, 2020-12-03:
Wait, misread. Reopen using the issue templates though because I need more information.

closed 2020-12-03 by SciresM

Gilgn, 2020-12-03:
> Wait, misread. Reopen using the issue templates though because I need more information.

How do I do that?

SciresM, 2020-12-03:
Make a new issue. Fill out the template instead of deleting it, like you did.

Gilgn, 2020-12-03:
> Make a new issue. Fill out the template instead of deleting it, like you did.

How do I do that? 😥

SciresM, 2020-12-03:
Use PC and not mobile.