Atmosphere-NX/Atmosphere #1228: Error when opening Tinfoil and game

Just8n12, 2020-12-03:
I currently updated lastest version of Armosphare but I can't run Tinfoil or any games on my switch

peterbgaming, 2020-12-03:
> I currently updated lastest version of Armosphare but I can't run Tinfoil or any games on my switch

Same problem here, i think, maybe, its a issue whit the sigpatches

Masamune3210, 2020-12-03:
Don't ask for help with piracy

BakedBit, 2020-12-03:
> I currently updated lastest version of Armosphare but I can't run Tinfoil or any games on my switch

I used the prerelease of of the cfw to downgrade back to 10.2.0 using choidujour so I can run my xci of smash and that worked for me

Just8n12, 2020-12-03:
@BakedBit How did you downgrade your firmware, I did not backup my Switch on 10.2.0

SciresM, 2020-12-03:
Piracy is not supported.

closed 2020-12-03 by SciresM