Atmosphere-NX/Atmosphere #1231: Switch can't turn on without SD card

Just8n12, 2020-12-03:
I updated my switch to 11.0 and use atmosphere 16.0
After i did it. I downgraded my switch to 10.2.0 used Choidujour

And now I still can turn on my switch with a SD card inside but when I take it out my switch can't turn on, it's like my switch missing the firmware and maybe firmware was moving to SD card, please help

masagrator, 2020-12-03:
Atmosphere doesn't support ChoiDojourNX and is against using it becsuse of nontrivial chance in data corruption.

fincs, 2020-12-03:
Friendly reminder that Daybreak exists.

Just8n12, 2020-12-03:
Any changes to fix it, now it's like my switch it's dead without sd card, btw I have my backup file but could it work?

masagrator, 2020-12-03:
If you have full backup, restoring it should help.

SciresM, 2020-12-03:
Atmosphere does not support booting without the ads card.

Also, you did not use the issue templates.

closed 2020-12-03 by SciresM

Just8n12, 2020-12-03:
i tried to restore my switch with a backup file and it seems like it is not working, my system firmware is disappeared. Please help me out, because if my SD card broke or something I won't able to use my switch anymore?

SciresM, 2020-12-03:
Fill out the issue templates. More context is needed.