Atmosphere-NX/Atmosphere #1237: Cannot Read Game Cards

KidLycan, 2020-12-03:
## Bug Report
### What's the issue you encountered?

When finishing the normal launch after doing the copy paste update method for the 11.0.0 switch update, after booting up as CFW, an error message stating that the game card couldn't be read. tried with Mario cart 8 Dlx and them Pokemon sword to see if the game cartage itself was the issue. It wasnt. I booted it to the Stock in the launch menu were CFW is and after it booted, the cartage was being properly read and able to be played!

### How can the issue be reproduced?

After copy-paste update, try booting up and launching CFW
once  you go to the main switch start up screen where you see the top 3 newest news, before a input is sent, a cannot read came card message pops up. And it happens after you insert a different and the same cartage. (and did the blow into the cartage gimmick too XD)

### System Firmware Version
 
Switch: 11.0.0
Hekate: 5.5.0
Nyx: v0.9.6
Atmosphère:  0.16.0 (https://github.com/Atmosphere-NX/Atmosphere/releases)

### Additional context?

I have one of the very older few models that can use the 10 pin way to get into the RCm mode.

All and all, its still able to play games cartage by going through the Stock launcher and able to play downloaded games from the CFW launcher. I just wanted ya to know about the issue.

switchtonite, 2020-12-03:
After i boot up cfw it loads up and all but when i click on the album menu it crashes.
Error Code 2144-0001 when going into the album folder.
---------------------------------------------------------------------------------------------------
Switch: 11.0.0
Hekate: 5.5.0
Nyx: v0.9.6
Atmosphère: 0.16.0 (https://github.com/Atmosphere-NX/Atmosphere/releases)

SciresM, 2020-12-04:
set nogc=0 in BCT.ini, or whatever the hekate equivalent is.

closed 2020-12-04 by SciresM

quizno4evah, 2020-12-04:
Can confirm this is happening on my switch as well with updates. 

Something I noticed using the NXDumptool v1.1.12 was that the 

"Error: unable to get root HFS0 header data!" pops up.

I double checked Hekate and turned of nogc to be sure, and this still appears. Any thoughts on what might be happening?

SciresM, 2020-12-04:
"disabled nogc" you probably did not. if using hekate, BCT.ini configuration is ignored. If using fusee primary, hekate_ipl configuration is ignored.

gamecards work fine on every confirmed console using correctly configured nogc = 0.

quizno4evah, 2020-12-04:
I found the BCT.ini file under SD:>atmosphere > config. The one I ended up changing earlier was the one under SD:>bootloader>hekate_ipl.ini which didn't do anything like Scires mentioned.

That fixed my problem and I can at least play from the cart now.