Atmosphere-NX/Atmosphere #1238: Black Screen after Atmosphere Launch

AniissB, 2020-12-03:
Good morning all

Following the release of Atmosphere 16.0 to run the switch 11.0 version, I have a black screen and I cannot access the main screen with the selection of games.

I have updated my files and my USB key with the fusee primary.bin file

The console is launched perfectly with the Atmosphere logo, recognizes the USB key but then black screen and the console crashes. You have to do a Hard reboot to restart the console

thank you for your help & your work !

gdicaterina, 2020-12-03:
Hi, how are you?
I was the same error, use the new hekate (release now) and my problema will solve.

Best regards!!

Grass0001, 2020-12-03:
i think the latest version of the build is broken or something, as i have installed atmosphere about ~6 hours ago and its been fine, but I've heard 3 people not have atmosphere boot in the last few hours

glosdaz, 2020-12-04:
I have the same issue booting from Hekate launch sysmmc black screen not even booing Atmos have to hard reset

SciresM, 2020-12-04:
Re-open using the issue templates, providing details about what custom sysmodules you use etc.

closed 2020-12-04 by SciresM