Atmosphere-NX/Atmosphere #1241: Black Screen after update Atmosphere for 11.0

AniissB, 2020-12-04:
## Bug Report

### What's the issue you encountered?

Following the release of Atmosphere 16.0 to run the switch 11.0 version, I have a black screen and I cannot access the main screen with the selection of games.

I have updated myself files and my USB key with the fusee primary.bin file
I followed the usual procedure and it always worked before but this time i have a Black Screen. 

### How can the issue be reproduced?
After updated my switch, the console is launched perfectly with the Atmosphere logo, recognizes the USB key but then black screen and the console crashes. You have to do a Hard reboot to restart the console

I can't do anything else than reboot the console

### Crash Report

No crash report is displayed. Only a black screen that lets me do nothing

### System Firmware Version

11.0 for my switch console
Atmosphere 0.16.0-p2 

### Environment?

- Atmosphere official 

### Additional context?

- My console is in airplane mode

Thank again for your help

SciresM, 2020-12-04:
Custom system modules? Post what's inside /atmosphere/contents.

jjorda, 2020-12-04:
I have the same problem but with the release version 0.16.0 not with the 0.16.0-p2. I am compiling 0.16.0-p2 to test.

AniissB, 2020-12-04:
Hi @SciresM 

No, it's not un custom system modules. Inside /atmosphere/contents i have this files : 
00FF0000636C6BFF
010000000007E51A
010000000000000D
010000000000002B
010000000000003C
0100000000000008
0100000000000032
0100000000000034
0100000000000036
0100000000000037
0100000000000042
0100000000000352
420000000000000E
690000000000000D
4200000000000010

Thank u for your feedback

masagrator, 2020-12-04:
I see 6 custom sysmodules.
Delete
00FF0000636C6BFF
010000000007E51A
0100000000000352
420000000000000E
690000000000000D
4200000000000010

And try again

AniissB, 2020-12-04:
@masagrator 

Ok i try it and tell you after this

Thx for your help

AniissB, 2020-12-04:
@masagrator 

Same problem, black Screen after Atmosphere Logo

SciresM, 2020-12-04:
To be sure: delete/rename the atmosphere folder temporarily, cleanly extract latest release zip?

THZoria, 2020-12-04:
I have exactly the same problem, clean installation from a freshly formatted SD. Black Screen when booting via primary fusee.
But works well with Hekate
Primary functional fuse after downgrade of the emumtm to 10.2.0 and downgrade of Atmosphere to 0.15.0

SciresM, 2020-12-04:
@THZoria that just sounds like misconfigured fusee to me.

THZoria, 2020-12-04:
> @THZoria that just sounds like misconfigured fusee to me.

I tested by configuring myself and using the preconfigured Deepsea pack. The problem occurs with both

SciresM, 2020-12-04:
"Configuring myself" this gives me extremely low confidence.

Did you test the obvious, which is to say "delete the atmosphere folder, extract latest zip with zero changes/additions"?

SciresM, 2020-12-04:
deepsea is also dumb as fuck/not supported.

THZoria, 2020-12-04:
> «Me configurer», cela me donne une confiance extrêmement faible.
> 
> Avez-vous testé l'évidence, c'est-à-dire "supprimer le dossier de l'atmosphère, extraire le dernier zip sans aucun changement / ajout"?

I am French I speak very bad English ^^
Yes I made the test to simply remove atmosphere (and at the same time bootloader for the new version of Hekate) and replaced everything with the .zip of the 0.16.0 available on github.

SciresM, 2020-12-04:
What logo screens do you see in what order?

THZoria, 2020-12-04:
> What logo screens do you see in what order?

the logo sept then atmosphere then black screen

SciresM, 2020-12-04:
You don't see the black/white atmosphere logo?

THZoria, 2020-12-04:
> You don't see the black/white atmosphere logo?

I have all the atmosphere logo, the only logo I don't have and the one for the Nintendo Switch at startup.

SciresM, 2020-12-04:
Emummc or sysmmc?

THZoria, 2020-12-04:
The two options

SciresM, 2020-12-04:
Which one are you using?

THZoria, 2020-12-04:
emummc

jcghizzo, 2020-12-04:
I have a similar problem, in my case the screen goes black after the nintendo switch logo appears.

i use emummc.

SciresM, 2020-12-04:
Does sysmmc work @THZoria?

THZoria, 2020-12-04:
> Does sysmmc work @THZoria?

After checking it works well in sysmmc

SciresM, 2020-12-04:
This actually sounds like an emummc issue then. @CTCaer insights?

AniissB, 2020-12-04:
@SciresM @THZoria 

I delete the Atmosphere, sept, Switch and HBmenu file and past the new version to be sure not to have any problem with the files.
It worked !! I was able to launch Atmosphere and find the main screen but unfortunately the games do not start anymore.

When I enter the Hbmenu, it asks me to install Daybreak and then no files work anymore (I can't find Goldleaf to install any more games and even my NSP folder is empty)

Do I need to install Daybreak?

SciresM, 2020-12-04:
Piracy is not supported.

closed 2020-12-04 by SciresM