Atmosphere-NX/Atmosphere #1243: Crash on launching Atmosphere 0.16.0

FerasArabiat, 2020-12-04:
## Bug Report

### What's the issue you encountered?
Hello,

I faced an issue when using Atmosphere 0.16.0. Initially, i used Atmosphere-updater to update to the latest version. I did all the updates in one go (Hekate, Atmosphere, Sigpatches and App, then restarted). I Saw the Sept Logo, and then the Atmosphere logo and after that the screen when to black.

I removed the SDCard, deleted Atmosphere folder, Sept folder and pasted in the content of the 0.16.0 release.  On boot, Sept was shown and atmosphere splash screen, after that the screen went to black and i got a fatal error.

```
Fatal Error Occured When Running Atmosphere
Program ID: 010041544d530000
Error Desc: std::abort() called (0xffe)
```

### How can the issue be reproduced?

1. Deleted old Sept Folder
2. Deleted old Atmosphere folder
3. Paste in content of 0.16.0 on root of SD
4. Boot into Atmosphere
5. Crash after atmosphere splash

### Crash Report

I have the bin format crash reports, gist does not allow those it seems, if there is a preferred way of adding those please let me know and I will provide them
</br>

### System Firmware Version

9.4.0 (i think but i'm going from memory)</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: official build 0.15.0 (atmosphere-0.16.0-prerelease-7336dc2b+hbl-2.4.0-prerelease+hbmenu-3.4.0-prerelease)
- Do you have additional kips or sysmodules you're loading: not sure.
- Homebrew software installed: N/A

### Additional context?

- Additional info about your environment: N/A

masagrator, 2020-12-04:
> 9.4.0 (i think but i'm going from memory)

That is a big miss. 9.4.0 doesn't exist

FerasArabiat, 2020-12-04:
> > 9.4.0 (i think but i'm going from memory)
> 
> That is a big miss. 9.4.0 doesn't exist

corrected, 9.2.0

FerasArabiat, 2020-12-04:
Seems that the issue is related to Mac (maybe Big Sur specifically)

I tried using version 0.15.0 with the mac and i faced the same issue. Tried it again using terminal instead of gui and the same issue remained.

I then move to windows, deleted atmosphere folder, sept folder and ran disk repair on the SDCard, after that i installed 0.15.0 and it worked fine. Repeated the steps on 0.16.0 and it also worked.

Seems that Mac is doing something strange to the file system on the SD Card.

SciresM, 2020-12-04:
>macOS

run archive bit fixer, macOS is known to fuck it up.

closed 2020-12-04 by SciresM