Atmosphere-NX/Atmosphere #1246: Black screen once I run cfw-sys (or stock)

psytech69, 2020-12-05:
## Bug Report

Black screen once I run cfw-sys (or stock)

### What's the issue you encountered?

On the launch screen of **HEKATE**, once I choose "cfw-sys" or "Stock", the SWITCH screen stays black...
No error message is displayed and I have to press OFF button for 10 seconds to unlock the SWITCH device.

### How can the issue be reproduced?

Today, I formatted SD card.
I got the new files from "https://www.sdsetup.com/".
I downloaded the package "Recommended Defaults" and "Minimal" to perform 2 separate tests.
The problem persists with either the "Recommended Defaults" or "Minimal" package. And to test the "Minimal" launch, I formatted SD card again and copied the "Minimal" package to the empty SD card.

### Crash Report

N/A

### System Firmware Version

Switch: 11.0.0

### Environment?
New files from   https://www.sdsetup.com/   
-> Atmosphere v0.16.0/2020.12.03.r1 / Hekate 5.5.0

### Additional context?

_**Before**_ the HOS 11.0.0 update, I had no problems running "cfw-sys" as usual.

SciresM, 2020-12-05:
Sounds like bad sept/etc.

Does issue occur booting using fusee-primary and latest atmosphere release zip with nothing else on SD?

psytech69, 2020-12-05:
I did a different test to start from scratch.

1. I erased my SD card.
2. I downloaded atmosphere-0.16.0-prerelease-7336dc2b+hbl-2.4.0-prerelease+hbmenu-3.4.0-prerelease.zip
3. I manually added the latest HEKATE version from : https://github.com/CTCaer/hekate/releases/tag/v5.5.0

In the bootloader\hekate_ipl.ini file I added the following lines because it was missing:

> [CFW (sysMMC)]
> emummc_force_disable=1
> fss0=atmosphere/fusee-secondary.bin
> icon=bootloader/res/icon_payload.bmp
> 
> [CFW (emuMMC)]
> emummcforce=1
> fss0=atmosphere/fusee-secondary.bin
> icon=bootloader/res/icon_payload.bmp
> 
> [Stock]
> emummc_force_disable=1
> fss0=atmosphere/fusee-secondary.bin
> icon=bootloader/res/icon_switch.bmp
> stock=1
> 
> [Fusee]
> icon=bootloader/res/icon_payload.bmp
> payload=bootloader/payloads/fusee-primary.bin

When I run TegraRcmGUI and Hekate, I still have the same problem with the choice of "CFW (sysMMC)" or "Stock".
**_-> persistent black screen._**

I didn't choose "CFW (emuMMC)" because I didn't do an emuMMC

And if I choose "Fusee" (**_I've never had to choose this launch_**), I get an error message that says : 
[ERROR] Fatal error: Failed to initialize the SD card! Press POWER to reboot

Namely that out of 20 launch attempts, I had a Seven error that appeared, but I didn't take a picture.
-

SciresM, 2020-12-05:
Does fusee succeed without launching via hekate? 20 failures in a row sounds to me like your SD card is dying or something similar.

Please report test results without using hekate at all.

SciresM, 2020-12-05:
Also, black screen on "stock" suggests pretty strongly to me your nand might be fucked, since atmosphere isn't even running in that case.

JteMonteEnDerby, 2020-12-06:
same for me when i launch custom firmware i got an error it say ERROR Fatal error : NXBOOT Unable to identify package1!

(press power to reboot)

JteMonteEnDerby, 2020-12-06:
so i try something, i formatted my sd card and i only reinstall athmosphere and i still get the error so i think the problem is with athmosphere

JteMonteEnDerby, 2020-12-06:
or maybe the new maj switch where they add switch online thing

SciresM, 2020-12-07:
Closing due to lack of response + seeming likelihood that it's SD card issues.

closed 2020-12-07 by SciresM

toastyishere21, 2021-06-24:
with booting its just empty black like the switch turned off im in 1 bit mode but as the guide said it would load into cfw but very slowly is there another way to boot cfw i was going to repair the sd card im not sure if this swich has been in cfw before