Atmosphere-NX/Atmosphere #1247: Crash when booting Atmosphere 0.16.0

sunyearbr, 2020-12-05:
## Bug Report

### What's the issue you encountered?

I updated the atmosphere to 0.16.0 and tryed to boot in Sysnand Atmosphere 

### How can the issue be reproduced?
Boot in Atmosphere sysnand 

### Crash Report
It's attached

### System Firmware Version

11.0.0

### Environment?

- bootloader: latest hekate (updated to support 11.0.0

[report_085c1f4a.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5647149/report_085c1f4a.zip)

HookedBehemoth, 2020-12-05:
Update [mission control ](https://github.com/ndeadly/MissionControl/releases/tag/v0.4.0)

sunyearbr, 2020-12-05:
I discovered the problem: I was using a sysmodule called "Mission Control". It's not updated to run on the new OFW 11.0.0. Sorry for the inconvenience.

closed 2020-12-05 by sunyearbr

ndeadly, 2020-12-05:
@sunyearbr it is https://github.com/ndeadly/MissionControl/releases/tag/v0.4.0

sunyearbr, 2020-12-07:
> @sunyearbr it is https://github.com/ndeadly/MissionControl/releases/tag/v0.4.0
Thanks @ndeadly and @HookedBehemoth, it is working now! :-)