Atmosphere-NX/Atmosphere #1249: I updated my firmware to 11.00

angel12455, 2020-12-05:
I updated my firmware to 11.00 by mistake so  I replaced all the necessary files in my sd card from the site and i got this

a fatal error occurred when running atmosphere
program ID 010000000000bd00
Error Desc: std: abort () Call (Oxffe)
Report saved to /atmosphere/fatal_errors/report_00000000086ba1b.bin

i just have to wait for the final files or did i did something wrong

bahrainiprog, 2020-12-05:
I did the same thing.
I updated the firmware to 11.0
Replaced all the Primary and Secondary files from the site.
Getting error messages when I start any application.

Masamune3210, 2020-12-05:
Please make a new issue and this time fill out the issue template

SciresM, 2020-12-05:
1) update mission control
2) use the issue templates

closed 2020-12-05 by SciresM