Atmosphere-NX/Atmosphere #1250: LayeredFS not working for one game

BigBoss0674, 2020-12-05:
## Bug Report

Game - Immortals Fenyx Rising, LayeredFS not working.

### What's the issue you encountered?

Exefs is ok, but when i putting files to romfs, the game just ignore all of them and "romfs_metadata.bin" file did not created.

### How can the issue be reproduced?

See above. Any game version.

### Crash Report

N/A

### System Firmware Version

10.2.0

### Environment?

Atmosphere 0.15.0 /  Hekate v5.3.4

### Additional context?

On Atmosphere 0.16.0 (pre-release) the same issue

SciresM, 2020-12-06:
Yeah, it uses a romfs open command (program index) which we don't actually support right now. Should probably add support.

LyuboA31, 2020-12-16:
i notice Xenoblade Chronicles 2 wont read exracted romfs files expect the libnx but if you extract the common.bdat from bf2.ard it will not read it at all it does however read the recompiled bf2.ard if this is something that can be fixed with atmosphere will be great

masagrator, 2020-12-16:
> i notice Xenoblade Chronicles 2 wont read exracted romfs files expect the libnx but if you extract the common.bdat from bf2.ard it will not read it at all it does however read the recompiled bf2.ard if this is something that can be fixed with atmosphere will be great

This is not the Atmosphere issue. Atmosphere won't help you replace and read files that game code doesn't expect. If you will unpack common.bdat from bf2.ard, game won't read that because it doesn't expect this file to be outside of bf2.ard. Original issue is completely different and valid case.

LyuboA31, 2020-12-16:
> > i notice Xenoblade Chronicles 2 wont read exracted romfs files expect the libnx but if you extract the common.bdat from bf2.ard it will not read it at all it does however read the recompiled bf2.ard if this is something that can be fixed with atmosphere will be great
> 
> This is not the Atmosphere issue. Atmosphere won't help you read files that game code doesn't expect. If you will unpack common.bdat from bf2.ard, game won't read that because it doesn't expect this file to be outside of bf2.ard. Original issue is completely different and valid case.

thx i thought so but wanted to ask someone more in the know

ash20493, 2020-12-22:
Didn't think it was worth creating a new ticket for, but layered fs seems to also not be working on origami king, have been trying to load the google translate patch and i have tried all suggestions i could find online

H4X0R46, 2021-01-14:
Just tried with Skyrim on Atmosphere 0.17.1 and it isn't working for me either. I tried atmosphere/contents AND atmosphere/titles. Nothing.

SciresM, 2021-01-14:
That's unrelated, and almost certainly a user issue. 

Skyrim doesn't use 7.x+ ProgramIndex apis.

H4X0R46, 2021-01-14:
> That's unrelated, and almost certainly a user issue.
> 
> Skyrim doesn't use 7.x+ ProgramIndex apis.

Okay, I'm glad to know it's user error at least. What's the path to use with atmosphere for LayeredFS?

SciresM, 2021-01-14:
no, doesn't matter, and skyrim doesn't use any different configuration from any other title. Just put mods in the correct program folder/romfs, as usual.

H4X0R46, 2021-01-14:
> no, doesn't matter, and skyrim doesn't use any different configuration from any other title. Just put mods in the correct program folder/romfs, as usual.

This is my path for Skyrim, no clue what's off here. Does this look right to you?

/atmosphere/contents/01000A10041EA000/romfs/

SciresM, 2021-01-14:
the atmosphere issue tracker is not a support forum.

please stop using unrelated issues for support.

Seek help on discord, in N-H or reswitched.

SciresM, 2021-06-29:
I finally looked at this game and it literally doesn't even use program index stuff.

commit a3dd445b (2021-06-29): fs.mitm: add OpenDataStorageWithProgramIndex support (theoretically, closes #1250)
closed 2021-06-29 by SciresM