Atmosphere-NX/Atmosphere #1253: Atmosphere crashes on boot

EmreTech, 2020-12-06:
### What's the issue you encountered?

When I attempt to boot Atmosphere from Hekate, it just crashes in the middle of boot.
I recently updated to Atmosphere version 0.16.0, and I updated to Firmware 11.0.0.
Before updating to Firmware 11.0.0, it worked fine with Firmware 10.2.0.

### How can the issue be reproduced?

You can reproduce this by updating to Firmware 11.0.0 after updating to Atmosphere 0.16.0.

### Crash Report

The crash report was saved to the fatal errors folder and I cannot upload it to gist.github.com.

### Environment?

I was using Hekate Ctcaer version 5.5.0.
I was using the official Atmosphere version 0.16.0.
I have sigpatches that are for Atmosphere 0.15.0 and SysDVR.

Nold360, 2020-12-06:
Same here. But I already had firmware 11.0.0 before launching atmosphere for the first time. 

[01607247135_0100000000001000.log](https://github.com/Atmosphere-NX/Atmosphere/files/5648633/01607247135_0100000000001000.log)

[01607247135_0100000000001000.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5648634/01607247135_0100000000001000.zip)

JteMonteEnDerby, 2020-12-06:
just wait until they find out i guess

Masamune3210, 2020-12-06:
Uninstall any sysmodules that dont come with atmosphere
Dont ask for help with sigpatches

EmreTech, 2020-12-06:
I wasn’t asking help with sigpatches, I was just saying that they are there.

fincs, 2020-12-06:
Piracy is not supported. Using sigpatches implies usage of piracy.

closed 2020-12-06 by fincs

EmreTech, 2020-12-06:
What the hell!

EmreTech, 2020-12-06:
You can’t just close this