Atmosphere-NX/Atmosphere #1255: Atmosphere doesn’t boot

EmreTech, 2020-12-06:
What's the issue you encountered?

When I attempt to boot Atmosphere from Hekate, it just crashes in the middle of boot.
I recently updated to Atmosphere version 0.16.0, and I updated to Firmware 11.0.0.
Before updating to Firmware 11.0.0, it worked fine with Firmware 10.2.0.

How can the issue be reproduced?

You can reproduce this by updating to Firmware 11.0.0 after updating to Atmosphere 0.16.0.

Crash Report

The crash report was saved to the fatal errors folder and I cannot upload it to gist.github.com.

Environment?

I was using Hekate Ctcaer version 5.5.0.
I was using the official Atmosphere version 0.16.0.
I have SysDVR.

EmreTech, 2020-12-06:
Here you go inconsiderate people. An updated bug report.

SciresM, 2020-12-06:
Does not reproduce here. Upload and link your crash report, as without it assistance can not be provided.

EmreTech, 2020-12-06:
Will do. I’ll try to upload soon.

EmreTech, 2020-12-06:
[Fatal Error Report.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5649186/Fatal.Error.Report.zip)

Alright, this does contain a bin file of the error because that's what it saved

SciresM, 2020-12-06:
Update mission control, which you did not mention you had, and which needs an update to be compatible with 0.16.0.

closed 2020-12-06 by SciresM

EmreTech, 2020-12-06:
Mission Control? Okay...

SciresM, 2020-12-06:
Yes, mission control is crashing due to 0.16.0 incompat.

EmreTech, 2020-12-06:
Oh. Thanks for the help!

ndeadly, 2020-12-06:
There must some AIO packages bundling MissionControl. There has been a large increase in users who deny running it lately despite the TID telling otherwise...

EmreTech, 2020-12-06:
AIO? Like AIO Switch Updater?

EmreTech, 2020-12-06:
I do have another error, but I'm just going to update all of my Homebrew applications that use USB and all my other Sysmodules

ndeadly, 2020-12-06:
> AIO? Like AIO Switch Updater?

All-In-One. I don't know what AIO switch updater does, but I'm talking about zip files put together by third parties that bundle Atmosphere along with other homebrew sysmodules and applications.

EmreTech, 2020-12-06:
Oh, those! I know of a website that kinda does that except you can customize it