Atmosphere-NX/Atmosphere #1256: Panic occurred while running Atmosphere: Title ID 010000000000002B

jlafaye, 2020-12-06:
## Bug Report

### What's the issue you encountered?

Atmosphere crashes with a panic error message

Title ID: 010000000000002B
Error: std::abort (0xFF3)

### How can the issue be reproduced?

The crash happens systematically.
I noticed it first with OFW 10.2.0 & atmosphere 0.15. Upgrading to 0.16 does not fix the issue.

On the same SD card, starting with a recreated emuMMC & a vanilla install of Atmosphere 0.16 and hekate_ctaer 5.5 allow the console to boot.

Restoring the (broken?) emuMMC leads to the panic error.
 
### Crash Report

### System Firmware Version

System version at time of crash 10.2.0
I am not able to start the CFM+emuMMC so I am not able to confirm if a console update is pending

### Environment?

- bootloader: hekate_5.5.0
- official build of atmosphere, downloaded from github
- Do you have additional kips or sysmodules you're loading: I don't think so because I started with a fresh install of atmosphere
- Homebrew software installed: the ones provided with atmosphere + aio-switch-updated (never started though)

### Additional context?

I am not able to mount the emu RAW partition with nyx.

Feel free to reach me if you need extra information.

jlafaye, 2020-12-06:
Crash report.

I was not able to upload the binary file so I have base64 encoded it.

https://gist.github.com/jlafaye/907a3614e8f4d80d0d48c36ea753ae18

EmreTech, 2020-12-06:
Try turning off Send Error Information in Settings? Same thing happened to me when I started out.

SciresM, 2020-12-06:
You have a corrupt system settings save. Delete the relevant save file and allow settings sysmodule to recreate it, or just do what you did and create a fresh/clean system state.

Either way, not an atmosphere bug.

closed 2020-12-06 by SciresM

jlafaye, 2020-12-07:
Thank you for the quick reply. How can deleting the system settings save be achieved ?

SciresM, 2020-12-07:
Pinging @Adubbz ^.

Adubbz, 2020-12-07:
You can use hekate's USB tools to mount your emuMMC's RAW GPP (Make sure you turn off Read-Only mode at the bottom). You can then use HacDiskMount's "Open physical drive" option on your PC (it'll show up as something like ``hekate SD GPP``) to mount the SYSTEM partition of your NAND. From here, go into the save folder and delete the file named ``8000000000000050``. Then, unmount the SYSTEM partition in HacDiskMount and then unmount the GPP in Windows and reboot your console.

Adubbz, 2020-12-07:
To add to this, you probably have a corrupt PRODINFO in your emuMMC. You can use HacDiskMount to restore that from a backup too.

jlafaye, 2020-12-10:
Thank you for your guidance. I deleted the files 8000000000000050 and 8000000000000052 as you mentioned but this resulted in a black screen.
Nevertheless, it pointed out that HacDiskMount can be used to mount the USER partition. Going this way, I was able to recover the saves and convert them into a suitable format for Edizon.

bejamartins, 2021-03-16:
@jlafaye Can you go though the process on how to convert the saves in the user partition for Edizon?

jlafaye, 2021-03-16:
I am pretty sure I started with @Adubbz's suggestion of mounting the firmware partition through HacDiskMount. I am not able to remember the exact procedure I followed but it involved no tool and a comparison of save filenames. I did not perform an actual conversion of the files. Sorry for not being able to provide more guidance.

bejamartins, 2021-03-16:
> I am pretty sure I started with @Adubbz's suggestion of mounting the firmware partition through HacDiskMount. I am not able to remember the exact procedure I followed but it involved no tool and a [...]

Thank you though. The mounting part I had understood, the conversion was where I was stuck at. Meanwhile, I discovered a tool called [Switch Army Knife](https://github.com/dezem/SAK) that did the trick for me. I used Checkpoint to restore it rather than EdiZon, can't get EdiZon to work with it.

Tidybloke, 2021-04-23:
@Adubbz 

I know this is closed but could you help?
I was directed here from my issue #1416 (which was closed before I could try this fix) and I followed your instructions but now I get a black screen. Not even the Atmosphere splash screen.

Is there anything you can think of that I could try to recover it?

Adubbz, 2021-04-23:
If that doesn't work I would suggest restoring a non-corrupt backup. https://suchmememanyskill.github.io/guides/unbrick/ is probably more comprehensive regarding what you should do in absence of one though.

ghaschel, 2021-08-12:
I renamed my atmosphere and sept folder, pasted a newest one from atmospheres github, pasted again the missioncontrol files, rebooted into hekate, ran the archive bit fixer because I was plugging in my sdcard to a MacBook (it creates a bunch of .files), and it worked

hero3210, 2023-06-17:
> You can use hekate's USB tools to mount your emuMMC's RAW GPP (Make sure you turn off Read-Only mode at the bottom). You can then use HacDiskMount's "Open physical drive" option on your PC (it'll show up as something like `hekate SD GPP`) to mount the SYSTEM partition of your NAND. From here, go into the save folder and delete the files named `8000000000000050` and `8000000000000052`. Then, unmount the SYSTEM partition in HacDiskMount and then unmount the GPP in Windows and reboot your console.

thanks for your helpful reply. 

If you can please edit your reply to say "emu RAW GPP" instead of "emuMMC's RAW GPP" to avoid confusions since this is what it's called now in Hekate. With a quick glance I actually edited the only one I saw with "MMC" which was the eMMC, I am not trying to blame you or anything, just a small suggestion to avoid future confusions. Thanks again.

For everyone coming here:
eMMC is the for the SysNand
emuMMC is the for the EmuNand

Chakyllfcu, 2024-08-18:
> You can use hekate's USB tools to mount your emuMMC's RAW GPP (Make sure you turn off Read-Only mode at the bottom). You can then use HacDiskMount's "Open physical drive" option on your PC (it'll [...]

When I try to mount the disk I got this: what can I do?

[image]