Atmosphere-NX/Atmosphere #1258: Black screen with close too fast any game and reopen nother game.

swat1990, 2020-12-07:
## Bug Report

### What's the issue you encountered?

Very good job everything works well except for a little thing I don't know if the other one has the same problem but when I close a game I sometimes have to wait a few seconds before opening another game or otherwise it gives me a black screen and nothing works anymore I have to turn off the console via the power button.

I have atmospher 0.16.0 and my switch is in 11.0.0 and I have the last sig-patch to install.
I have the impression that it may be a question of stabilizing, but I am not sure to check it please and thank you for the very good work.

### How can the issue be reproduced?

This make with any game just have to open any game close this game and re-open another game for get black screen after reboot the same game work perfect if i wait few sec after close a game i never get this black screen, before in atmospher 0.15.0 i never have this bug.

### Crash Report

N/A
### System Firmware Version

11.0.0

### Environment?
On switch home menu.

### Additional context?
N/A

SciresM, 2020-12-07:
You did not provide context about custom system modules.

What do you have installed?

Does the issue reproduce with clean atmosphere with absolutely nothing else?

swat1990, 2020-12-07:
I have test with only atmosphere and i have the bug again for custom system modules  i dont have any custom system module.

SciresM, 2020-12-08:
Send a screenshot of /atmosphere/contents folder?

swat1990, 2020-12-08:
[image]

SciresM, 2020-12-08:
"no custom system modules"

You have 10 custom system modules. 

Why did you not tell the truth earlier?

Remove custom system modules, and test then if the issue still reproduces.

swat1990, 2020-12-08:
Sorry I thought he had more and I just saw the bug this product in fact when I start tinfoil then after closing if I open a jex that's where I have a black screen even without module.

swat1990, 2020-12-08:
[image]

SciresM, 2020-12-08:
"tinfoil" 

piracy is not supported.

Sounds like this isn't an atmosphere issue.

closed 2020-12-08 by SciresM

swat1990, 2020-12-08:
ok thank for the info