Atmosphere-NX/Atmosphere #1260: No save data with Atmosphere 0.16.0

SJMazzella3380, 2020-12-07:
## Bug Report

Since updating to 0.16.0, my save data is missing.  The only way to get it back is to shut down the switch and boot into OFW.
If I boot into CFW, I have to redownload all of my games and I cannot access my save files.

All of my games are officially purchased through the E-Shop,  I only use CFW for the cheat functionality.

So currently I have to download my games from the E-Shop twice, once for OFW and once for CFW.  Then I have two sets of save files.  One for OFW and one for CFW.

Masamune3210, 2020-12-07:
Make sure you arent using emunand or anything like that. Make sure save redirection is turned off

And please reopen using issue template if that doesnt solve anything

SJMazzella3380, 2020-12-07:
Reopened using the template and updated the problem I am having.  Thank you

closed 2020-12-07 by SJMazzella3380