Atmosphere-NX/Atmosphere #1261: No Game Data/Save Data Atmosphere 0.16.0

SJMazzella3380, 2020-12-07:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br> NO
[ - software explicitly distributed for piracy, etc ]</br>NO
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>SD card not showing data while in CFW
[ Did you make any changes related to Atmosphère itself? ]</br>Updated to 0.16.0
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br> While in CFW, SD card shows there is data used, but I cannot access it, I have to redownload all of my EShop games and I have no save data access either.  Once I redownload a game, I can play it normally and have to start a new save file.

If I power down and reboot into OFW, then I have access to all the games that I have and all the original save files.
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

11.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee
- Official release or unofficial build:  latest relase on github atmosphere 0.16.0
  - [ Official release version x.x.x (or) unofficial build ] 
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

Since updating to 0.16.0, my save data is missing. The only way to get it back is to shut down the switch and boot into OFW.
If I boot into CFW, I have to redownload all of my games and I cannot access my save files.

All of my games are officially purchased through the E-Shop, I only use CFW for the cheat functionality.

So currently I have to download my games from the E-Shop twice, once for OFW and once for CFW. Then I have two sets of save files. One for OFW and one for CFW.

- Additional info about your environment:
I can only get the CFW to work by loading the fusee-primary.bin payload

If I try to load the Atmosphere FSSO EmuMMC , I get the error message 
Unkown pkg1 version.  HOS version not supported! 
Or emuMMC corrupt!
trying backup bootloader... found pkg1(20201030110855').
Unkown pkg1 version.  HOS version not supported! 
Or emuMMC corrupt!
Failed to launch HOS!
Press any key...

If I try to load the Stock SYS, I get the error message 
Unkown pkg1 version.  HOS version not supported! trying backup bootloader... found pkg1(20201030110855').
Unkown pkg1 version.  HOS version not supported! Failed to launch HOS!
Press any key...

If I try to load the Atmosphere FSSO SYS, I get the error message 
Unkown pkg1 version.  HOS version not supported! trying backup bootloader... found pkg1(20201030110855').
Unkown pkg1 version.  HOS version not supported! Failed to launch HOS!
Press any key...

- [ Any other information relevant to your issue. ]

SciresM, 2020-12-07:
What's the version string in system settings?

M3ssor3m, 2020-12-08:
I have a similar issue with this pre-release version: 11.0.0|AMS 0.16.0|S
When will a stable version be released, any ideas?
Some apps are unresponsive now, they haven't been updated yet though.
Other apps are still working great like FTPD etc
Are they any sigpacthes available yet for the new version?

SciresM, 2020-12-09:
Closing because this guy is almost certainly just accidentally using emummc and he's not being responsive.

closed 2020-12-09 by SciresM