Atmosphere-NX/Atmosphere #1265: Error/Potential brick after downgrade from 11.0.0 to 10.2.0

xdCallMeCooper, 2020-12-08:
After downgrading from 11.0 to 10.2, this error code pops up on my switch after booting into atmosphere - 

Error Code: 2002-2634 (0x149402)
Program: 010000000000001F
Firmware: 10.2.0 (Atmosphère 0.15.0-master-1c71d12d)

An error has occurred.

It's been doing this for a day now and I fear that my Switch is completely bricked. When I try to turn it on normally without booting into atmosphere, all that appears is just a black screen. It's stuck there and I'm worried.

masagrator, 2020-12-08:
What tool you have used to downgrade FW?

HookedBehemoth, 2020-12-08:
Eject your gamecard and try again.

You updated the gamecard controller on 11.0.0 and FS now fails to handle it.

xdCallMeCooper, 2020-12-08:
> What tool you have used to downgrade FW?

Daybreak.

xdCallMeCooper, 2020-12-08:
> Eject your gamecard and try again.
> 
> 
> 
> You updated the gamecard controller on 11.0.0 and FS now fails to handle it.

Gamecard?

HookedBehemoth, 2020-12-08:
[image]

xdCallMeCooper, 2020-12-08:
> [image]
> 
> 

I don't even have a gamecard in my switch...

SciresM, 2020-12-08:
Yes you do.

In any event, set nogc=1 in BCT.ini, or whatever the hekate equivalent is.

closed 2020-12-08 by SciresM