Atmosphere-NX/Atmosphere #1266: Black screen on boot - SX Core V1

ajbritain4, 2020-12-09:
**What's the issue you encountered?**

Issue: black screen on boot, or fails to wake from sleep, depending on which bootloader used, on V1 patched Switch with SX Core chip, 11.0 sys and emuMMC. 

Fusee-primary always results in a black screen on boot regardless of how it’s loaded. Fusee-secondary either produces a black screen on boot, or boots to Atmosphere but fails to wake from sleep. Same behaviour in both stock sysMMC and CFW emuMMC. 

No changes made to Atmosphere - added to SD card exactly as latest release.

**How can the issue be reproduced?**

V1 patched switch with SX Core chip:

SX OS 3.1.0 bootloader (after Cleanup) - fusee-primary boots to black screen. Loading Hekate then Atmosphere results in a black screen using FSS0 or fuse-primary.

SX Gear boot.dat (loads payload directly bypassing SXOS bootloader) - fuses-primary boots to black screen. Fusee-secondary boots to OFW sysMMC and CFW emuMMC successfully but then hangs to a blank screen and requires a hard reset when sleep mode activated in both.

**Crash Report**
No crash report found.

**System Firmware Version**
11.0

**Environment?**
* What bootloader (fusèe, hekate, etc) was Atmosphère launched by: 
Fusee-primary - black screen on boot in all situations. Hekate - black screen or failure to wake from sleep as per details above. 
* Official release or unofficial build:
    * Official pre-release 16.0 and Deepsea 1.9.0
* Do you have additional kips or sysmodules you're loading: no
* Homebrew software installed: no

SciresM, 2020-12-09:
Not supported yet. Wait for 0.17.0.

closed 2020-12-09 by SciresM

ajbritain4, 2020-12-09:
Thank you - wasn’t aware of that sorry - thought that was just Mariko/Lite units

krishjuh, 2021-07-03:
I have this same issue on version: 0.19.1 does anybody know if or when this is going to be fixed? After the switch falls in sleep than i can’t get into Atmosphere. When i shutdown the switch and reboot i can get into Atmosphere.

ajbritain4, 2021-07-03:
> I have this same issue on version: 0.19.1 does anybody know if or when this is going to be fixed? After the switch falls in sleep than i can’t get into Atmosphere. When i shutdown the switch and reboot i can get into Atmosphere.

It’s already been fixed in 0.19.2 I think? So just update your atmosphere and you should be good, it’s been working totally fine for me since

ajbritain4, 2021-07-03:
> I have this same issue on version: 0.19.1 does anybody know if or when this is going to be fixed? After the switch falls in sleep than i can’t get into Atmosphere. When i shutdown the switch and reboot i can get into Atmosphere.

I’d try updating first of all to the latest Atmosphere and see how you get on