Atmosphere-NX/Atmosphere #1267: A fatal error occurred... v0.16.0

oreillygshane, 2020-12-09:
## Bug Report

### What's the issue you encountered?

Updated to v0.16.0
Then updated to 11.0.0
After Hekate boot:

"
A fatal error occurred when running Atmosphere
Program ID: 010000....bd00
Error Desc: std:::abort() called (0xffe)
Report saved to /at..././report_000...8b36bf5.bin

Press POWER..
"

### How can the issue be reproduced?

[ * ]</br>
Boot using v0.16.0 with 11.0.0</br>

### Crash Report

[report_0000000008b36bf5.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5669196/report_0000000008b36bf5.bin.zip)

### System Firmware Version

11.0.0</br>
</br>
</br>
</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  - hekate_ctcaer_5.5.0.bin
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - Latest Deepsea, but then overwrote everything with the latest Atmosphere due to a fusee_secondary issue (their filename was "fusee-secondary.bin.new" with the main .bin missing
- Do you have additional kips or sysmodules you're loading:
     - Everything under Deepsea
- Homebrew software installed: [ * ]

acnh-backup-restore
AIO-atmosphere-updater
appstore
Awoo-Installer
Checkpoint
ChoiDujourNXv102
DeepSea-Toolbox
DeepSea-Updater
EdiZon
ftpd
Goldleaf
JKSV
ldnmitm_config
Lockpick
NSAInstaller
NX-Shell
NXDumpTool
nxmtp
OpenLara
sigpatch-updater
Snake
Switch_themes_Installer
switch-cheats-updater

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

ndeadly, 2020-12-09:
You need to update MissionControl to the [latest version](https://github.com/ndeadly/MissionControl/releases/tag/v0.4.0)

oreillygshane, 2020-12-09:
AMAZING, that did the trick. Closing, thanks!!

closed 2020-12-09 by oreillygshane