Atmosphere-NX/Atmosphere #1268: Reboot to Payload (nro) not working!

pedroverri, 2020-12-10:
I don't know why but since I updated Hekate and Atmosphere to work with the firmware 11.0 when I launch the NRO application to reboot to payload it just put my Switch into RCM Mode. This is specially weird because when I used TegraRcmGUI to inject other payloads (like TegraExplorer for example) when I reboot to the payload it goes to Hekate 5.5.0 no problem.

SciresM, 2020-12-10:
1) you didn't use the issue templates
2) known issue with the nro, will be fixed in 0.17.0.

closed 2020-12-10 by SciresM

pedroverri, 2020-12-10:
Just noticed something. When I start the application but my Switch is not plugged to the computer the following message appears:

"
"Failed to mount SD!
Missing LP0 (sleep mode) lib!
Missing or old Minerva lib!
Update bootloaderfolder!
Press any key..."

And then it goes to Hekate CTCaerMod

pedroverri, 2020-12-10:
> 1. you didn't use the issue templates
> 2. known issue with the nro, will be fixed in 0.17.0.

Oh ok sorry. This was my first time reporting anything. I was confused. Thanks!