Atmosphere-NX/Atmosphere #1269: ipatched with SX Core - Boots stock/cfw fine but no boot with SX Core removed

yoboto, 2020-12-11:
What's the issue you encountered?
This seems different compared to the already closed issue #1266

With SX Core installed on iPatched switch, I booted atmosphere on FW V9.0.0 sysnand and installed v11.0.0 FW update via daybreak, factory defaults. After, Booting stock FW sysnand from SX boot menu works fine and booting stock and atmosphere CFW via hekate works fine too. but with EMMC removed and put back in original slot and SX taken out of the equation, pressing the power button or promting to boot with the cable just results in a black screen, behaviour is similar to a FW/Fuse mismatch. If I plug the EMMC back in via SX-Core it once again boots up stock/CFW fine, doing a cleanup from the SX boot menu makes no difference.

Am I missing something obvious?

Burnt Fuses count in Hekate displays as 14 - 9 which is correct based on the FW version as far as i know

How can the issue be reproduced?
Update ipatched switch from FW 9.0.0 to 11.0.0 with daybreak -sysnand with SX core installed then boot without sx core

System Firmware Version
11.0.0

What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Hekate
Official release or unofficial build:
Official
0.16.0 pre - opt in

SciresM, 2020-12-11:
I am highly confident that this is not an atmosphere issue. Seek support with spacecraft-nx, maybe?

SciresM, 2020-12-11:
("the console doesn't boot without the modchip present" doesn't seem like anything to do with atmosphere, so far as I know).

closed 2020-12-11 by SciresM

yoboto, 2020-12-11:
Ok, thanks for the help :)

Didn't even know about  spacecraft-nx will check it out, thanks.

btw is it possible this is an issue with Boot0 not being updated for some reason due to modchip interference?

SciresM, 2020-12-11:
Not just possible, probable ^^

yoboto, 2020-12-11:
Just add for posterity for anyone in the same boat

NXNandManager doesn't seem to be able to interpret the  FW ver for v11 boot0 files and just shows NA

dumped boot0 from hekate on said console

generating the stripped boot0 with emmchaccgen for FW11 and comparing to the new Hekate dump in Hex editor and it doesn't seem reveal any SX modification/interference afaict and the two files are comparable - aside from the obvious

Assuming the boot0 file generated by emmchacgen is valid for fw 11 then it would seem boot0 is not the issue in this case.... at least based on my very limited understanding.

Forgot to mention, I was using SX 3.0.4 boot.dat if that makes any difference

Guess i'll repeat the update process in daybreak but with the spacecraft-nx firmware instead and see if there's a change and report back