Atmosphere-NX/Atmosphere #1273: Atmosphere won't boot (Stuck on Nintendo Switch Logo) after update on 11.01 firmware and 0.16.1

joeyg8292002, 2020-12-13:
After, I update 11.01 firmware and Atmosphere 0.16.1 then when I try to launch Atmosphere, the regular Atmosphere screen comes up for a few seconds, then it goes black, then the Switch leaves RCM (my pc makes that "device disconnected" sound), then the Nintendo Switch logo freezes and shows up on the screen and stays like that forever. I try to fix everything on the SD Card but everything was fine but there was a problem is Nintendo Switch logo freezes.

Is there anything to fix this?

SciresM, 2020-12-13:
Reopen using the issue templates.

closed 2020-12-13 by SciresM

joeyg8292002, 2020-12-24:
Thanks, and I use hekate 5.51.bin but when I use Atmosphere, Atmosphere won't boot (Stuck on Nintendo Switch Logo) and Nintendo Switch logo freezes and shows up on the screen and stays like that forever. Nintendo Switch logo freezes but I updated from 0.16.1 to 0.16.2 in Atmosphere 0.16.2 and 11.01 firmware update and I try to fix it but I use NAND Backup and I fix it but when I use Atmosphere, Nintendo Switch logo still freezes and It can be fusee-primary.bin or Atmosphere is broken? and I have to wait for the next update and next firmware is update to 11.02 firmware update on Nintendo Switch.