Atmosphere-NX/Atmosphere #1276: 0.16.1 Not Booting

TwistedTempest, 2020-12-14:
Upon booting fusee-primary on 11.0.1 and reaching the Nintendo Switch logo, I am immediately prompted with this screen:
A fatal error has occurred when running Atmosphère.
Program ID: 010000000000bd00
Error Desc: std::abort() called (0xffe)
Report saved to /atmosphere/fatal_errors/report_000000000879bb06.bin

SciresM, 2020-12-14:
1) you did not use the issue templates
2) update mission control

closed 2020-12-14 by SciresM

TwistedTempest, 2020-12-14:
what are the issue templates?

SciresM, 2020-12-14:
When you make an issue, the text box is prepopulated with a form that instructs you to fill it out.

Users who do not fill out the template are ignored, because I have no interest in providing support to people who do not do me the courtesy of giving me the information I need.