Atmosphere-NX/Atmosphere #1277: Fatal error 0000000005766c2b.bin

sallmorais, 2020-12-15:
I'm having trouble unlocking the switch.

when I turn on the atmosphere the following error appears:

A fatal error occurred when running atmosphere.
program ID: 010041544d530000
error desc: std: :abort() called (0xffe)
report saved to /atmosphere/fatal_errors/report_0000000005766c2b.bin

the atmosphere directory may improperly have archive
bits set. please try running an archive bit fixer tool
(for example, the one in Hekate)

Press POWER to REBOOT

HookedBehemoth, 2020-12-15:
1. You didn't fill out the issue template
2. Take the advice it gives you to heart and run the archive bit fixer in [Hekate](https://github.com/CTCaer/hekate/releases)

SciresM, 2020-12-15:
Behemoth said what I would have said.

closed 2020-12-15 by SciresM

sallmorais, 2020-12-15:
Sorry for my ignorance but I still don't know what to do?