Atmosphere-NX/Atmosphere #1279: Unable to read Card Games + Launch Installed NSPs.

AlbertoCoder97, 2020-12-16:
## Bug Report

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
After updating Atmosphere with Kosmos' All In One Package (I also have Hekate) and Launching fusee-primary.bin from Hekate Menu (I also tried fusee-secondary), as soon as I tried to put a Game cartridge in, the switch gave an error about being unable to read the cartridge. Also I am not able to launch installed NSPs/NSz with Awoo Installer (I have the Atmosphere's sigpatches, I downloaded them from Atmosphere's Pre-Release and put them in the SD Card)

[ Did you make any changes related to Atmosphère itself? ]</br>
No.

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
1 - Update to the latest Armosphere version in pre-release with Kosmos' All In One Package.
2 - Insert Game Cartridge.

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
https://gist.github.com/AlbertoCoder97/b59c735e29a452d54648618a1d4bb8fa

### System Firmware Version

11.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Hekate Menu -> Launched fusee-primary payload.
- Official release or unofficial build: 
  - [ Official release version 0.16.1]

### Additional Informations
I'm not an expert of CFW/Homebrew but I never ran into this issue before. I updated Hekate + Atmosphere using the Kosmos' All In One Package.

masagrator, 2020-12-16:
1. Kosmos is dead, don't use it
2. Atmosphere should be updated only when HOS is shutdown.
3. Piracy is not supported.

fincs, 2020-12-16:
Piracy is not supported.

closed 2020-12-16 by fincs