Atmosphere-NX/Atmosphere #1281: Impossible to read the game card

manuelmanuel, 2020-12-16:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ AMS 0.16.0]</br>
[ N/A]</br>
[ N/A]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ With Atmosphere 16.0.0 the game card slot does not work]</br>
[ NO]</br>
[ N/A ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Insert a game card ]</br>

### Crash Report

[ N/A]</br>
[ N/A]</br>
[ N/A]</br>

### System Firmware Version

11.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2020-12-16:
Set nogc=0 in BCT.ini.

Also, you didn't actually fill out the template.

closed 2020-12-16 by SciresM

manuelmanuel, 2020-12-17:
> Set nogc=0 in BCT.ini.
> 
> Also, you didn't actually fill out the template.

Thank You!!!