Atmosphere-NX/Atmosphere #1282: Launching Atmosphere ends up with a black screen

PixyGixy, 2020-12-16:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

 After injecting fusee-primary.bin (latest, from /releases) the screen is black

### How can the issue be reproduced?

1. Put fusee-primary.bin on the rcm loader
2.Put the latest atmosphere download on the sd card
3. inject it and shows up with a black screen 

### Crash Report

Nothing in the file.

### System Firmware Version

11.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

Used hekate and the stand alone rcmloader way of injecting it, but still it ends up with a black screen

### Additional context?

i have reformatted, used a archive bit fixer but still has a black screen.

manuelmanuel, 2020-12-16:
Have you a Card Game inside the slot?

PixyGixy, 2020-12-16:
> Have you a Card Game inside the slot?

Uh yes i do, I'll be surprised if that's the problem and nope that isn't the problem

SciresM, 2020-12-16:
"black screen"

Please list every logo you see, in order.

Does issue occur using plain atmosphere with no custom sysmodules? Send a screenshot of your /atmosphere/contents folder.

PixyGixy, 2020-12-16:
> "black screen"
> 
> Please list every logo you see, in order.
> 
> Does issue occur using plain atmosphere with no custom sysmodules? Send a screenshot of your /atmosphere/contents folder.

[image]

by black screen i mean like when i inject it literally nothing shows up like no logos or anything but i know for sure that it has been injected

SciresM, 2020-12-16:
That sounds highly suspect. Can you test injecting without your loader dongle from PC, with latest release zip fully extracted to SD card?

What happens if you attempt to boot via hekate?

PixyGixy, 2020-12-16:
> That sounds highly suspect. Can you test injecting without your loader dongle from PC, with latest release zip fully extracted to SD card?
> 
> What happens if you attempt to boot via hekate?

i have tried that but it still ends up with a black screen and when i used hekate and clicked on payloads then chose the fusee-primary.bin it still has a black screen

SciresM, 2020-12-16:
Does boot via hekate not using fusee-primary succeed? Honestly this sounds like you have broken sept or something.

PixyGixy, 2020-12-16:
> Does boot via hekate not using fusee-primary succeed? Honestly this sounds like you have broken sept or something.

Nope it does not, but how would you fix a broken sept

ttajaf68, 2020-12-16:
im following and have the same atmosphere/contents and same black screen

ttajaf68, 2020-12-16:
i have used the files on the NH switch guide to set up

SciresM, 2020-12-16:
My best guess is that there's something fucked in your emmc/sd card. I flatly can't reproduce this.

My advice is to focus on booting via hekate and not fusee-primary, and then ask @CTCaer for help getting hekate to work -- I'm not skilled enough with bootloader debug to really assist you.

ttajaf68, 2020-12-16:
thanks for the help

SciresM, 2020-12-16:
No idea. Make sure it's formatted as fat32, though.

mox512, 2020-12-17:
Hi have same here.
Shows black-white A logo and stuck on loading. 
Nothing in file. 100% FAT32 SD Card reformatted from Hekate.
Checked archive bit - it's not it.

SciresM, 2020-12-17:
@mox512 if you see logos at all, your issue is different from this guy's. send screenshot of your /atmosphere/contents, so that we can be sure you have no custom sysmodules interfering.

ttajaf68, 2020-12-17:
when launching atmosphere the first time it showed the hekate logo then the sept logo only for a fraction of a second before it went black. The times after that the sept screen didnt show up.

ttajaf68, 2020-12-17:
Just a note to Pixygixy - atmosphere 16.1 pre-release worked fine for me im on 11.0.0

JteMonteEnDerby, 2020-12-17:
athmosphere is bugged asf

JteMonteEnDerby, 2020-12-17:
you just can't launch it with only athmosphere files

Masamune3210, 2020-12-17:
Just because it doesn't work for you, doesn't mean it doesn't work for everyone....

JteMonteEnDerby, 2020-12-17:
so why it don't work :c

JteMonteEnDerby, 2020-12-17:
a day it work and a day it don't

mox512, 2020-12-17:
> @mox512 if you see logos at all, your issue is different from this guy's. send screenshot of your /atmosphere/contents, so that we can be sure you have no custom sysmodules interfering.

Sure, here you go.
[image]

SciresM, 2020-12-17:
@mox512 you're missing almost all of atmosphere. Extract latest release zip in full and you should be fine.

closed 2020-12-19 by PixyGixy