Atmosphere-NX/Atmosphere #1283: Cannot Start Certain Games when in Atmosphere 1.6.1

Sukurudo, 2020-12-16:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ After successfully launching into atmosphere, I can start all but two titles. One a Cartridge game: Zelda:BOW, and one a digital download, Animal Crossing, NH. When I attempt to start the game, the initial Nintendo switch black screen shows, and it hangs there indefinitely. I can press the home button and X out of the title. Both games run normally when running Switch in normal mode.]</br>
[ No, other than updating to 16.1]</br>
[ N/A ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Boot Sustem into Hekate, and starup Atmostphere in SysNand; Attempt to run either Zelda:BOTW or Animal Crossing ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

0.16.1</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Hekate 5.5.1
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ Edzion, goldleaf, retroarch ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2020-12-16:
What custom sysmodules are you using? You did not fill out that part of that template.

Post screenshot of your /atmosphere/contents folder.

Sukurudo, 2020-12-16:
[image]
[image]

Here are the screenshots of my contents and my atmosphere folders. I have no custom sysmodules

SciresM, 2020-12-16:
"I have no custmo system modules"

"Screenshot shows you have 6 custom system modules"

Delete your custom system modules. Does the issue occur after you do that?

Sukurudo, 2020-12-16:
Oh, I didnt know that those were the sys modules. should I just replace the folder with a fresh one from Atmosphere?

SciresM, 2020-12-16:
Yes, and test if that works, please ^^

Sukurudo, 2020-12-16:
That was it! Thank you; Ill have to figure out what those were; the only thing I remember installing was Edzion overlay.

closed 2020-12-16 by SciresM