Atmosphere-NX/Atmosphere #1288: The game card could not be read.

chinos0228, 2020-12-19:
Bug Report
The game card could not be read.

What's the issue you encountered?
All my game card could not be read on Atmosphere 0.16.0 (pre-release), 0.16.1 (pre-release) and 0.16.2

How can the issue be reproduced?
Insert any game card and the error window pop up.
***
The game card could not be read.
Please remove and reinsert it.
If this message appears repeatedly, it may mean that the game card or the console is malfunctioning.
Visit nintendo support website for more information.
***

Crash Report
N/A

System Firmware Version & Environment
11.0.1 & Atmosphere 00.16.2 / Hekate v5.5.1

Additional context?
All the game cards are working properly before system update to 11.0.0.
All the game cards are working properly on my other consoles, and I assume my game game is not broken.
All the game cards are working properly without Atmosphere on the same console, I assume my console hardware is not broken.

SciresM, 2020-12-19:
1) you didn't use the issue templates correctly

2) copy BCT.ini from /config_templates/ to /config/ and add the nogc = 0 line as mentioned in INI comment.

closed 2020-12-19 by SciresM

fincs, 2020-12-19:
Add nogc = 0 to atmosphere/config/BCT.ini.

dazzaXx, 2020-12-21:
Heya, i've also ran into this gamecard issue, even when having the nogc line added into BCT.ini.
Does it go under the stratosphere section or stage1? I have it under stratosphere with no change to the error.

SciresM, 2020-12-21:
@dazzaXx look at/edit the template. It goes in the stratosphere section. Ensure that there's correct whitespace on the line.

SciresM, 2020-12-21:
Also, be sure that you're editing a copied INI inside /config/, and not the template itself.

dazzaXx, 2020-12-21:
So i did put it into the correct area, i have made sure that it is where it should be, both in the .ini and directory still with no luck, i've even put it on various different lines.
[image]

SciresM, 2020-12-21:
Where is the actual file that you're editing located?

dazzaXx, 2020-12-21:
/atmosphere/config

SciresM, 2020-12-21:
Firmware version?

dazzaXx, 2020-12-21:
11.0.1

SciresM, 2020-12-21:
Are you actually launching via fusee-primary? Are you sure you're not using hekate?

dazzaXx, 2020-12-21:
Oh.. does using hekate skip loading the configs?

SciresM, 2020-12-21:
Hekate is a fusee-alternative. It has its own completely separate config unrelated to fusee's, and does not use fusee's config.

dazzaXx, 2020-12-21:
Ahh, i did not know that. I have launched fusee-primary and the issue has gone. Sorry for wasting your time, now i know. 😅