Atmosphere-NX/Atmosphere #1289: Fatal Error on Atmosphere 1.16.2 after updating to firmware 11.0.1

Fer3243, 2020-12-19:
## Bug Report

After updating to firmware 11.0.1, when launching atmosphere 1.16.2/1.16.1 (i tried both), it gives me a fatal error. 

### What's the issue you encountered?

When launching atmosphere, Switch crashes after Switch logo with fatal error. "Title ID: 0100000000000004" and "Error desc: std::abort() called (0xffe)"

### How can the issue be reproduced?

I dont know if this will reproduce the issue but what i did was, Dump 11.0.1 firmware files with Tegraexplorer for Daybreak, then use my keys and EmmcHaccGen 2.2.2 (ChoiDujour110 didnt work to me) to convert it, and then transfer the files to my emunand.

### Crash Report

[report_000000000a2c9eab.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5720095/report_000000000a2c9eab.zip)

### System Firmware Version

11.0.1

### Environment?

I tried to launch it with hekate and fusee-primary.bin
I used Atmosphere 1.16.2
Homebrew software installed: [ Sysdvr, Edizon, Triplayer and Missioncontrol ]

### Additional context?

i updated from 11.0.0 and maybe EmmcHaccGen 2.2.2 doesnt support 11.0.1

SciresM, 2020-12-19:
Wait for mission control update.

SciresM, 2020-12-19:
@ndeadly we should probably figure something out so that this stops happening every time I release an atmosphere update.

closed 2020-12-19 by SciresM

ndeadly, 2020-12-20:
Fwiw the latest MissionControl release (0.4.1) boots just fine on 0.16.2 for me without the need to update anything. I think this user is just running an old version that wouldn't have worked with 0.16.1 either.