Atmosphere-NX/Atmosphere #1294: SaltyNX crashes on M.16.2, not crashing on 0.16.2

masagrator, 2020-12-20:
## Bug Report

### What's the issue you encountered?

I have V1.</br>
SaltyNX 0.4.0a was working without issue on M.15.0. </br>
Updated Atmosphere today to M.16.2 to investigate some issue. And it turned out no game now wants to load. </br>
Heap size is set correctly, nintendo logo disappears and after few seconds any game is crashing.</br>
Issue doesn't exist on 0.16.2, also when injecting plugins is disabled the issue doesn't exist on M.16.2</br>

### How can the issue be reproduced?</br>

1. Install SaltyNX 0.4.0a and Atmosphere M.16.2 (don't know if issue exists on newer firmwares), download NX-FPS as plugin example</br>
2. Run any game</br>

### Crash Report</br>

[01608472125_010039400e8d6000.log](https://github.com/Atmosphere-NX/Atmosphere/files/5720991/01608472125_010039400e8d6000.log)</br>

### System Firmware Version</br>

10.2.0</br>

### Environment?</br>

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: </br>
 fusee</br></br>
- Official release or unofficial build:</br>
 official M.16.2</br></br>
- Do you have additional kips or sysmodules you're loading: </br>
no kips;</br>
SaltyNX, Tesla, sys-clk, Noexs. </br>
Deleted everything except SaltyNX and issue still exists.</br></br>
- Homebrew software installed: </br>
I don't think it matters in this case, but I have few of my test homebrew, Daybreak, Checkpoint, Cheats Updater, hbappstore, moonlight, mtp-server-nx, NX-Shell, nxdumptool, nxmtp, PPSSPP, reboot to payload, Reset Parental Controls, RetroArch, ReverseNX-Tool, SaltyNX-Tool</br>

### Additional context?</br>

It happens for both game cartridges and installed games with and without updates. </br>

I will understand if this report will be closed because of how SaltyNX works. But if mesosphere behaves differently than official kernel, I think it's worth investigating what is happening.

Skonikol, 2020-12-20:
> ### How can the issue be reproduced?
> 1. Install SaltyNX 0.4.0a and Atmosphere M.16.2 ( + FW 10.2.0, don't know if issue exists on newer firmwares), download NX-FPS as plugin example
> 2. Run any game

I've tested a few games, everything works fine...

[image]

[image]

masagrator, 2020-12-20:
Can you provide a list of games? Or you have checked one of games I have listed and it works for you? I want to be as much be certain as possible that this issue is on my side.

Skonikol, 2020-12-20:
> Can you provide a list of games? Or you have checked one of games I have listed and it works for you? I want to be as much be certain as possible that this issue is on my side.

Ori 1 & 2
Catherine FB
RE 1
Mario Tennis Ases
Memoranda
FF 12 ZAge

I can check LA Noire but later, since I don't have it installed and the internet is bad. Unfortunately, I don't have a Witcher3.

Skonikol, 2020-12-20:
> you have checked one of games I have listed and it works for you?

Witcher is OK, Little Busters! too

[image]

masagrator, 2020-12-20:
Thanks for info. So this is something from my side. I will try to clean up shit and try again. Closing issue.

closed 2020-12-20 by masagrator