Atmosphere-NX/Atmosphere #1297: Mariko System save 8000000000000120

zal16, 2020-12-20:
I have a Mariko Switch with corrupted SYSTEM partition, but i tried to rebuid this partition with suchmememanyskill's Emmchaccgen, but he generates a 800000000000120 save for Erista consoles

would you like to get this file for me?

Ia located in SYSTEM/save/800000000000120

SciresM, 2020-12-20:
This is not what the issue tracker is for.

closed 2020-12-20 by SciresM

zal16, 2020-12-20:
how can i get this?

shchmue, 2020-12-20:
this is an emmchaccgen issue, not an atmosphere one, and i doubt there's any difference between this file on a particular firmware on erista vs mariko so your problems probably lie elsewhere

zal16, 2020-12-20:
i already talked to him, but he has no mariko console to compare.

when I turn on the console without this save file, it generates one, but it has a different size.

so I would like to have a file like this from a functional mariko console, just to compare and, if possible, help the suchmememanyskill

Masamune3210, 2020-12-21:
That's a real pickle, but this still isn't an issue with atmos. Your better off going somewhere else for help with that unless you can manage to prove somehow that atmos is related