Atmosphere-NX/Atmosphere #1299: failed to apply noldrchk

Themainman1360, 2020-12-22:
I was on the previous build, before this one and I just updated. Whenever I try to boot emmummc it gives me this error and puts me back into the hekate menu.

SciresM, 2020-12-22:
piracy is not supported, and this is not the hekate repo.

closed 2020-12-22 by SciresM

Themainman1360, 2020-12-22:
I'm sorry, I didn't know this was related to piracy, this only happened
when I updated to your latest release. If you could point me in the right
direction I would appreciate it.

icyph3r, 2024-02-23:
I am also getting this error, not related to piracy (unless the guides i have been following online have pointed me in the wrong direction). Any help resolving this would be amazing! I installed a homebrew app called 'sigpatch-updater', this was the last thing I did before getting this error, maybe it has something to do with that. Help appreciated!

Masamune3210, 2024-02-24:
N O      P I R A C Y
[image]

icyph3r, 2024-02-24:
Like I said before, (keep in mind I know nothing about switch modding) I was following a guide and appear to have been pointed in the wrong direction. I was not trying to pirate games, I was trying to install the version of a game directly from a game cart without it updating/installing the latest version. I fixed the `failed to apply noldrchk` error by starting from scratch and following a different guide.

SciresM, 2024-02-24:
That is piracy, which is not supported.