Atmosphere-NX/Atmosphere #1303: Atmosphere crashes after a few seconds in

Kiruhi, 2020-12-26:
## Bug Report

### What's the issue you encountered?

I just updated the firmware to 11.0.1 and Atmosphere to 0.16.2. It launches flawlessy, but after ~10 seconds in, it crashes, whether I open HBChannel, a game, or nothing

### How can the issue be reproduced?

Just waiting in the Home menu

### Crash Report

[01608990926_00ff0000a53bb665.log](https://gist.github.com/Kiruhi/74fa6a8cd95bd6a22edfa6c0b72cc6d0#file-01608990926_00ff0000a53bb665-log)

### System Firmware Version

11.0.1

### Environment?

fusèe-primary.bin provided in this same github. Happens everywhere, even in Home menu

### Additional context?

Nothing really

Kiruhi, 2020-12-26:
I got to fix it by removing atmosphere and sept folders from the sd root and pasting in a clean copy of both

Skonikol, 2020-12-26:
just update SysDVR

closed 2020-12-27 by SciresM