Atmosphere-NX/Atmosphere #1305: Atmosphere crashes when a gamecard is inserted

Metamorphie, 2020-12-27:
## Bug Report

### What's the issue you encountered?

I updated Atmosphere and Hekate to the latest versions (0.16.2 and 5.5.1 respectively)
Also, based on some solutions on here, people recommended adding nogc = 0 to the BTC.ini file, which I tried and it did not work. It's entirely possible that I did not do it right. 
[image]

### How can the issue be reproduced?

Updating to the latest Atmosphere and Hekate versions and inserting a game card.

### Crash Report

https://gist.github.com/Metamorphie/0e43b8caf8a3e00792666c7734fa9c6d

### System Firmware Version

11.0.1</br>
Although in CFW, it says I'm on 10.0.4. 

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee-primary
- Official release or unofficial build: 0.16.2
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: 

### Additional context?

Sorry if I filled in something wrong here, I'm completely new to this stuff. Again, in CFW it says I'm on version 10.0.4 while on Horizon I'm on 11.0.1. I'm not really sure why.
I was able to use cartridges fine on previous versions, but now it's not working. Any help would be appreciated.

SciresM, 2020-12-27:
If using hekate, BCT.ini isn't used, edit the hekate specific configuration for nogc.

thank you very much for using the issue templates.

Metamorphie, 2020-12-27:
I am using hekate. So, how would I go about editing the hekate configuration then? You're right in that I don't see BCT.ini in the bootloader folder. This is what I see in the bootloader folder:

[image]

Would I change the hekate_ipl.ini file then? There's a line for "autonogc," so would I change that to 0? 
[image]
Thanks for your fast reply.

Skonikol, 2020-12-27:
You need to update emummc too to version 11.0.1. Now you have updated the FW of the cartridge slot and it remains inoperable on a lower firmware.

Metamorphie, 2020-12-27:
That makes sense! I can't believe I didn't think of that. I'll try it and see if it works.

Metamorphie, 2020-12-27:
Okay... I have never updated emuMMC before, so I'm following this guide: 
https://nh-server.github.io/switch-guide/extras/updating/#updating-emummc-by-dumping-an-updated-firmware-from-your-sysmmc
It's the same one I used to hack my switch. It's going well but this is my issue: On the Updating emuMMC section, step 2 is:
Once booted, hold R while launching a game to boot into the homebrew menu.
I can't do this. Now it says that the game card could not be read (which is better than it crashing entirely). I do have digital games, but it says that they're not on the console and need to be downloaded. Is it safe to do this on CFW? I don't want to risk a ban from Nintendo.
So, should I set autonogc = 0 now to fix it? Sorry for the questions, but I just want to be careful and do this right.

Skonikol, 2020-12-27:
> Okay...

To log in to hbmenu, use applet mode (hold down R to start album, in some settings, you can simply log in to album without R)

Metamorphie, 2020-12-27:
Just finished updating and everything works great! Thank you both for your help and patience!

closed 2020-12-27 by Metamorphie