Atmosphere-NX/Atmosphere #1310: Atmosphere boot problem, please help

ChillyWilly2, 2021-01-01:
## Bug Report

### What's the issue you encountered?
I was updating my nintendo switch to the newest firmware, I updated atmosphere to the newest version and then I updated my nintendo switch to firmware 11.0.1 with ChoidujourNX when the installation was done I rebooted and got the fatal error screen

It says
Program ID: 010000000000bd00
Error Desc: std::abort() called (0xffe)
Report Saved to /atmosphere/fatal_errors/report_report_0000000008a7ac3e.bin
Press POWER to reboot

I don't know how to get back to rcm or do anything for that matter cause it doesn't give me the option to do anything but reboot. When it reboots it reboots straight to atmosphere and the error repeats every time.

### How can the issue be reproduced?

I do not know, I cant get out of the fatal error screen to test anything else. It just keeps rebooting and giving the error screen forever

### Crash Report

N/A

### System Firmware Version

11.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build:  Official 0.16.2
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: N/A

### Additional context?

- Additional info about your environment: The only relevant information is that I updated my switch to 11.0.1 and it caused the error. I had it run normally on firmware version 10.2.0

shchmue, 2021-01-01:
update or disable MissionControl

ChillyWilly2, 2021-01-01:
> update or disable MissionControl

where and how do I do that?

shchmue, 2021-01-01:
download and replace the version on sd or else remove the file `atmosphere/contents/010000000000bd00/flags/boot2.flag`

closed 2021-01-01 by SciresM