Atmosphere-NX/Atmosphere #1312: game card not reading issue is back

butiuae, 2021-01-03:
after the recent update my card reader stopped working and the already proposed fix with the BCT.ini and Hekate_ipl.ini doesn't work completely. The error message is removed but when you run a game it loads for a few seconds then fails with an error message. Downloaded games are not affected in any way even when using nogc=0

Skonikol, 2021-01-03:
Since you have not used the template for issue, it is very difficult to give you the right advice.

butiuae, 2021-01-03:
What do you mean by template?

Skonikol, 2021-01-03:
> What do you mean by template?

https://github.com/Atmosphere-NX/Atmosphere/issues/new?template=bug_report.md

butiuae, 2021-01-03:
## bug report
**what is the issue encountered?**
game card could not be read

**How can the issue be reproduced?**
I just updated atmosphere to version 0.16.2

**Crash Report**
N/A

**system firmware version**
11.0.1

**environment**
atmosphere version 0.16.2
hekate  version 5.5.1

**additional context**
I use the website sdsetup.com when updating using there recommended Pre-configured bundles.
This is a similar issue to issue#794 but the solution no longer works.
https://github.com/Atmosphere-NX/Atmosphere/issues/794

Skonikol, 2021-01-03:
Do you use emummc? What is the configuration in your hekate_ipl.ini ?

butiuae, 2021-01-03:
i use sysmmc.

**hekate_ipl.ini**
[config]
autoboot=0
autoboot_list=0
autohosoff=1
autonogc=0
backlight=100
bootwait=3
updater2p=0

[CFW (sysMMC)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
icon=bootloader/res/icon_payload.bmp

[CFW (emuMMC)]
emummcforce=1
fss0=atmosphere/fusee-secondary.bin
icon=bootloader/res/icon_payload.bmp

[Stock]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
icon=bootloader/res/icon_switch.bmp
stock=1

[Fusee]
icon=bootloader/res/icon_payload.bmp
payload=bootloader/payloads/fusee-primary.bin

Skonikol, 2021-01-03:
You need to show what error appears. Perhaps this has nothing to do with nogc.

butiuae, 2021-01-03:
if nogc=1 then the standard error
[image]

but if nogc=0 then the game will load for a second then:
[image]

Skonikol, 2021-01-03:
If you run this configuration, how does the GC run behave?

[Fusee]
icon=bootloader/res/icon_payload.bmp
payload=bootloader/payloads/fusee-primary.bin

butiuae, 2021-01-03:
it behaves the same way

butiuae, 2021-01-03:
i checked all configs and it looks like i can play the GC if i run the stock config but any other way it dosn't work

butiuae, 2021-01-03:
its fine now that i am sure that that the actual switch is fine and that i have a way to play physical games i will wait for the next update and do a clean intall since i didn't have this problem with previous versions it might work on the newer versions thanks for your help

closed 2021-01-03 by butiuae