Atmosphere-NX/Atmosphere #1316: Random crash when installing software to NAND

micktu, 2021-01-06:
Methods employed:

* DBI "Install to NAND."
* HOS "Move software to internal storage"
* Game Card Installer "Nand Install"

A crash stack appears a few gigabytes into transfer pretty consistently.

[image]

SciresM, 2021-01-06:
Piracy is not supported.

`HOS "Move software to internal storage"` --> Known issue with file-based emummc, not an atmosphere issue.

closed 2021-01-06 by SciresM

micktu, 2021-01-06:
I understand that piracy is a thing that should not exist. I was just having fun backing up my legal copies. Thank you, I will look around for the Emummc issue, Apparently, it affects all off the NAND encoding methods.

SciresM, 2021-01-06:
"backing up legal copies"

piracy, not supported.