Atmosphere-NX/Atmosphere #1317: blank_prodinfo_sysmmc=1 is hanging out system

rashevskyv, 2021-01-06:
## Bug Report

### What's the issue you encountered?

`blank_prodinfo_sysmmc=1` on `exosphere.ini` make loading hangs on joycon logo on Mariko 

With `blank_prodinfo_sysmmc=0` system loads fine. 

Tested on `atmosphere-0.17.0-master-e53b9cc5+hbl-2.4.0+hbmenu-3.4.0` loaded directly from `fuse-primary.bin`, that was launched from spacecraft-NX

### How can the issue be reproduced?

* Unpack `atmosphere-0.17.0-master-e53b9cc5+hbl-2.4.0+hbmenu-3.4.0` to SD 
* Put `fuse-primary.bin` to SD and rename it to payload.bin
* Launch console. 

**Expected result:**
Atmosphere loads to HOS 

**Actual result:**
Atmosphere will freeze on Joy-con Logo (last logo before HOS usually starts)

### System Firmware Version

11.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
    - fusèe
- Official release or unofficial build: 
  - atmosphere-0.17.0-master-e53b9cc5
- Do you have additional kips or sysmodules you're loading:
  - no
- Homebrew software installed: 
  - daybreak

### Additional context?

exosphere.ini contents: 

```[exosphere]
debugmode=1
debugmode_user=0
disable_user_exception_handlers=0
enable_user_pmu_access=0
blank_prodinfo_sysmmc=1
blank_prodinfo_emummc=1
allow_writing_to_cal_sysmmc=0
log_port=0
log_baud_rate=115200
log_inverted=0```

SciresM, 2021-01-06:
Prodinfo blanking is a direct 1:1 copy of the original incognito logic, made before Mariko released. It's probably not compatible with newer prodinfo formats.

Not really an atmosphere issue since same problem will occur just using incognito.

I would accept a PR/instructions on how to blank newer format prodinfo, if the research is done by someone.

closed 2021-01-06 by SciresM

rashevskyv, 2021-01-06:
NxNandManager blanked prodinfo without issues

SciresM, 2021-01-06:
That does not sound like a PR/instructions to me.

rashevskyv, 2021-01-06:
I tried to make incognito in emunand with blawar's incognito.nro 1.5 and it is works on Switch lite with 11.0.1, so it is not a logic issue

SciresM, 2021-01-06:
sysmmc != emummc, and presumably you have different e.g. network settings which could have impacts. I stand by my statement.

rashevskyv, 2021-01-06:
I tried on sysnand, then make emunand from this sysnand, tried on this emunand with Mesosphere and without it - same result. 
Then I tried incognito.nro and all works. 
I can make a video if you think that my experiment was not clear enought

SciresM, 2021-01-06:
The atmosphere blanking code is a duplicate of the original incognito code, produced using the original incognito code as a reference.

Given I'm aware of format issues with newer prodinfo, I find it substantially more likely that your testing methodology is flawed.

rashevskyv, 2021-01-06:
I admit that I can doing something wrong
I would appreciate it if you could recommend a testing methodology that suits you.

H4X0R46, 2021-01-07:
This is a real problem, seems like a LOT of Mariko users are reporting exosphere making the system hang at boot, emummc as well!

SciresM, 2021-01-07:
It's of course a real thing that occurs, that's not disputed.

I welcome a PR or instructions.

rashevskyv, 2021-01-07:
Maybe you should reopen an issue?

H4X0R46, 2021-01-07:
A PR? What do you mean? Some people are using incognito and it seems to work in place of exosphere. I myself am unsure....... Some say normal incognito is hacky and bricks sometimes 👀

SciresM, 2021-01-07:
Why would I re-open the issue? That would imply I'm putting development effort into it. 

I've talked to parties who will look into it, I think, but it's out of my hands.

I'll get a PR (pull request) or instructions, and then I'll merge the fix or implement the fix as instructed.

If neither of those happens, the issue will remain unresolved.

Fwiw, atmosphere's blanking is based on the tinfoil incognito implementation, which recalculates CRCs. The incognito NRO does not.

There are literally 5+ ways that people "blank" prodinfo without a standard.

I implemented the most common one at the time the feature was added.

Given the lack of a standard, and the issue being that the current blanking (crc recalculation) causing issues with newer prodinfo formats, I'm waiting for someone who actually researches the prodinfo format to instruct me on what to do, which I think is quite reasonable.

H4X0R46, 2021-01-07:
To be fair, you said that there's other parties looking into it, and that's all I wanted to hear. I'm content with that. Also, I messaged you on Twitter a very simple yes or no question, whenever you get a moment. Same name as on here.

SciresM, 2021-01-07:
Ah, I didn't see your twitter DM. Your question elsewhere is innocuous enough for public commentary:

Q: "Hey SciresM, have a quick question for you. I'm about to set up atmos on my Mariko V2 and wanted to know if daybreak has been tested and confirmed safe and working for a Mariko emummc?"

A: Yes, daybreak is tested and working on Mariko. You should use it to update offline. If you're not using spacecraft-nx, you'll get a longer boot time than usual the first update after daybreak is used, but there's no risk of a brick or anything.

H4X0R46, 2021-01-07:
Awesome! Thanks scires! Gonna head to bed, have a good night yourself 👍

H4X0R46, 2021-01-07:
Question, what's this line for? I've heard some say this needs to also be enabled for blanking? Idk, haven't messed with it.

allow_writing_to_cal_sysmmc=0

Masamune3210, 2021-01-08:
As far as I remember, it allows or disallows writing to the calibration data for the system

H4X0R46, 2021-01-12:
Hey scires, I see that the fix has been added, when are you planning a new build? I'm excited to finally have atmos on my dumb Mariko! Lol

rashevskyv, 2021-01-12:
You can built it by yourself actualy

H4X0R46, 2021-01-12:
> […](#)

Yeah, being GitHub and an open source project I know. I'm not experienced with how git works enough to compile from it.

EmreTech, 2021-01-13:
If you build Atmosphere yourself, this issue wouldn't be a problem.

H4X0R46, 2021-01-13:
> If you build Atmosphere yourself, this issue wouldn't be a problem.

iF You BUild AtmOsPheRE yOURSelf, THIS IsSuE WOuldn'T Be A pROBlEm.

SciresM, 2021-01-13:
@H4X0R46 don't be a shithead.

The fix will make it into Atmosphere 0.17.1, which will be released at an unspecified time in the near-ish future.

If you're desperate for the fix before then, compile a build of atmosphere with the fix included in the meantime.