Atmosphere-NX/Atmosphere #1319: Failure after boot.kip

Arch9SK7, 2021-01-06:
## Bug Report

### What's the issue you encountered?

 
Trying to boot Atmosphere CFWSysnand from Hekate (not from the payloads menu) via SX lite mod chip and i get a failure after boot.kip it does a power off reset combo into normal OFW. No error messages.</br>
None. It is a clean version.</br>

### How can the issue be reproduced?

</br>
so this has been tested using SX OS bootloader off the chip on SX lite and spacecraft-NX bootloader on chip. I dropped the newest version of atmosphere and Hekate onto SD. I launch using hekate payload. If i press payloads then choose Fusee-primary.bin i can successfully launch Atmosphère but here is my catch.
I like to use NOEXE and i ran this on a prebuild of atmo for mariko just fine before. So to do this i am needing to boot CFWSys with no changes i choose launch and CFWSys from hekate. 

I then get Atmos B&W splash then the nintendo splash and HOS launches. I check version settings and its vanilla HOS nothing with atmosphere.</br>

### Crash Report

This issue doesnt provide me with a report or fatal crash. </br>

### System Firmware Version

11.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate 5.5.2
- Official release or unofficial build: 
  - [Official release version 0.17.0 ]
- Do you have additional kips or sysmodules you're loading: Yes i had Noexs.kip1 (failed with and without this)
- Homebrew software installed: [ just the default from Official Atmosphère build 0.17.0]

### Additional context?

- Additional info about your environment:
- This was tested using SX LITE on SX bootloader and Spacecraft-NX bootloader for the chipset.

SciresM, 2021-01-07:
This sounds like your hekate boot config is bad and you're only partially launching atmosphere. Does the issue occur launching via fusee-primary?

Arch9SK7, 2021-01-07:
> This sounds like your hekate boot config is bad and you're only partially launching atmosphere. Does the issue occur launching via fusee-primary?

Nope it doesnt happen just with fusee. And i had Ctcaer check my config and he saw no issues with it so im not too sure.

SciresM, 2021-01-07:
I don't know then. If it doesn't occur with fusee, it's definitely not an atmosphere issue, though...

SciresM, 2021-01-07:
You can send me your config on discord, as well, and I can see if I spot anything.

closed 2021-01-07 by SciresM