Atmosphere-NX/Atmosphere #1323: Daybreak's handling of AutoRCM?

SwiftMove, 2021-01-07:
I don't see any documentation or mention of how Daybreak handles AutoRCM. With ChoiDujourNX, there was an option to Enable or Disable AutoRCM when Upgrading/Downgrading. Daybreak doesn't have the same option, which brings the question of how does Daybreak handle AutoRCM? Does it always Enable AutoRCM by default? Or does it just ignore the first byte of data (where AutoRCM lives) when updating the four BCT's, thereby leaving whatever AutoRCM option the user has chosen?

SciresM, 2021-01-07:
Daybreak is a system updater tool. System updates are unrelated to autorcm, so it's not talked about in the documentation.

It doesn't touch autorcm at all. If the user has autorcm and uses daybreak, they will have autorcm after. If the user does not have autorcm and uses daybreak, they will not have autorcm after.

closed 2021-01-07 by SciresM

SwiftMove, 2021-01-07:
That's what I wanted to know. Thanks for the swift response.