Atmosphere-NX/Atmosphere #1327: Atmosphère Fatal Report (v1.1):  0x1BA69 (2105-0221) Can somebody help?

samielis, 2021-01-08:
Hello Everyone!
Glad to be able to finally use atmosphere on my mariko, thank you so much for your work.
I am having trouble with the firmware, it crashes when trying to start.
If somebody has an idea what the problem is, i would be really happy.
Thank you!

## Bug Report

### What's the issue you encountered?

I installed Spacecraft-NX, Hekate and Atmosphere on my Mariko Switch.
It booted proberly. Afterwards i took the MicroSD Card out and added some additional homebrew and some Savegames for Checkpoint.
After reinserting and booting Atmosphere through Hekate, i get this Fatal Error.

### How can the issue be reproduced?

Happens every time when trying to boot Atmosphere.

### Crash Report

https://gist.github.com/samielis/5e2aba7fe0ac143c8ae3431c582b7527

### System Firmware Version

10.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:  
- Spacecraft-NX 0.1.0 -> Hekate_ctcaer_5.5.2 -> Atmosphere 0.17.0
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Checkpoint, EdiZon, tinfoil

### Additional context?

HookedBehemoth, 2021-01-08:
Check the system_settings.ini file in ``atmosphere/configs``
The error code suggests that it is malformed.

0x1BA69 | 105 | 221 | Setting group name has zero length

samielis, 2021-01-09:
That was the fix. Thank you very much!

closed 2021-01-09 by SciresM