Atmosphere-NX/Atmosphere #1329: A fatal error ocurred when moving a large size game into system storage

oBattleTurtleo, 2021-01-09:
## Bug Report

### What's the issue you encountered?
- A fatal error

### How can the issue be reproduced?
- Moving a large size game (5GB) into system storage.

### Crash Report
- [Here](https://gist.github.com/oBattleTurtleo/02b9ba021351119d3f94ff3253c08d50)

### System Firmware Version
- Nintendo Switch Mariko on 10.2.0

### Environment?
- Hekate bootloader
- Official release version [Atmosphère 0.17.0](https://github.com/Atmosphere-NX/Atmosphere/releases/tag/0.17.0)
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: 
  - [DeepSea](https://github.com/Team-Neptune/DeepSea), Atmosphère is excluded
  - [ReverseNX-RT](https://github.com/masagrator/ReverseNX-RT)

### Additional context?
- "N/A"

SciresM, 2021-01-09:
Known issue in file based emummc, not an issue with atmosphere itself.

closed 2021-01-09 by SciresM

Miroxyz, 2021-01-11:
@SciresM I am facing the same problem. Will the problem be solved if I change to SD Partition based emummc?