Atmosphere-NX/Atmosphere #1332: (v1 Erista patched) with blackscreen on start Atmosphere.

jhonathas, 2021-01-14:
## Bug Report

The Hekate opens in a good way on the chip with Erista, but when I try to load the Atmosphere, it does not start and the black screen appears and is locked.

### What's the issue you encountered?

I try to start Atmosphere using hekate or directly and the screen goes black and freezes. I tried to start from the SxCore SXOS payload menu and likewise crashes. At some point, after I gave SXOS a cleanup, I managed to get in, but there were many errors and then I couldn't do it anymore.

Using the RetroReloaded package, I can enter Atmosphere, but I end up losing interest because their package is very modified. Modify the entire menu.

### How can the issue be reproduced?

I tried using emunand hidden and also files, in both cases the problem persists.

### Crash Report

NA

### System Firmware Version

Sysnand and Emunand on version 11.0.0

### Environment?

I tried Pure Atmosphere with SXOS.
I tried Pure Atmosphere with sxgear's boot.dat
I tried via hekate
In the tests I had the cleanest possible environment, without homebrews that could cause problems and I also used the version without mothosphere to tests

### Additional context?

NA

SciresM, 2021-01-14:
If atmosphere boots when using some package, that reads to me like your boot configuration for atmosphere is just fucked in some way.

Cleanly extract latest release zip with no add-ons/etc. If using chip, use spacecraft-nx or similar. Problem will be resolved, and if it's not you should post screenshots of your clean configs/a short video of the boot process.

closed 2021-01-14 by SciresM

jhonathas, 2021-01-14:
I came to use a clean package, free of add-ons, but it still wasn't. I am in a group with several people with the same problem. I'll send some screens for you to see.

I thought about using spacecraft-nx, is it worth it then? Do you indicate others besides spacecraft-nx?

SciresM, 2021-01-14:
nothing else is officially supported.

asbeserra, 2021-01-14:
V1 Erista patched no boot, Blackscreen when start Atmosphere

SciresM, 2021-01-14:
As I said, `post screenshots of your clean configs/a short video of the boot process`.

Issue's gonna end up being something in your boot configuration.

rashevskyv, 2021-01-14:
Make cleanup first, then go to payloads and tun fuse-primary to boot atmosphere. 
You should make cleanup every time you power on your console. 

After cleanup you can install spacecraft to your modchip.

SciresM, 2021-01-14:
"Cleanup" not necessary since 0.17.0.

rashevskyv, 2021-01-14:
Without cleanup i got boot parse error. After cleanup all fine. 
Did not try 0.17.1 yet. Working on 0.17.0

SciresM, 2021-01-14:
Oh, right, we detect BCT2 is necessary via my public key. That's pretty amusing, wontfix.

jhonathas, 2021-01-14:
Is it something that can be resolved in a future version?

jhonathas, 2021-01-15:
> If atmosphere boots when using some package, that reads to me like your boot configuration for atmosphere is just fucked in some way.
>
> Cleanly extract latest release zip with no add-ons/etc. If [...]

Hello, @SciresM I created a video to show the problem, I hope it helps.

https://youtu.be/qhqeHxCx9KE

SciresM, 2021-01-15:
Yeah, as mentioned above, use spacecraft-nx or cleanup your boot0.

Boot via non-spacecraft is not actually supported.

jhonathas, 2021-01-15:
How to clean boot0? Is Clean on the sx-core menu? If so, I already did, but it also didn't solve it. Maybe the solution is really spacecraft

SciresM, 2021-01-15:
"I already did" needs to be done every boot. If you do it, and then reboot to use fusee, it's the same as not having cleaned up at all.

If you want atmosphere to work, use spacecraft.