Atmosphere-NX/Atmosphere #1333: Crash on boot after updating System FW and atmosphere

TheRadziu, 2021-01-14:
## Bug Report

### What's the issue you encountered?

Updated System to latest FW at this time, Updated Atmosphere & Hekate to their latest versions, tried to boot via hekate or via fusee, resulted in a system Crash during Nintendo Switch Screen.
Also removed all directories but EmuMMC and Nintendo before copying fresh files from the zip(s)

### Crash Report

```
Atmosphère Fatal Report (v1.1):
Result:                          0x149402 (2002-2634)

Program ID:                      010000000000001f
Process Name:                    ns
Firmware:                        9.1.0 (Atmosphère 0.17.1-master-49f07b06)
General Purpose Registers:
Start Address:                   0000004523200000
Stack Trace:
        ReturnAddress[00]:       000000452333985c
        ReturnAddress[01]:       0000004523338fe4
        ReturnAddress[02]:       00000045233390b8
        ReturnAddress[03]:       000000452323223c
        ReturnAddress[04]:       0000004523232058
        ReturnAddress[05]:       0000004523232128
        ReturnAddress[06]:       00000045232ba328
        ReturnAddress[07]:       000000452326a3dc
        ReturnAddress[08]:       0000004523208744
        ReturnAddress[09]:       0000004523309ec8
        ReturnAddress[10]:       000000452330ce24
        ReturnAddress[11]:       0000000000000000
        ReturnAddress[12]:       0000000000000000
        ReturnAddress[13]:       0000000000000000
        ReturnAddress[14]:       0000000000000000
        ReturnAddress[15]:       0000000000000000
        ReturnAddress[16]:       0000000000000000
        ReturnAddress[17]:       0000000000000000
        ReturnAddress[18]:       0000000000000000
        ReturnAddress[19]:       0000000000000000
        ReturnAddress[20]:       0000000000000000
        ReturnAddress[21]:       0000000000000000
        ReturnAddress[22]:       0000000000000000
        ReturnAddress[23]:       0000000000000000
        ReturnAddress[24]:       0000000000000000
        ReturnAddress[25]:       0000000000000000
        ReturnAddress[26]:       0000000000000000
        ReturnAddress[27]:       0000000000000000
        ReturnAddress[28]:       0000000000000000
        ReturnAddress[29]:       0000000000000000
        ReturnAddress[30]:       0000000000000000
        ReturnAddress[31]:       0000000000000000
```

### System Firmware Version

9.1.0 EmuMMC
11.0.1 System</br>

### Environment?

- Latest Hekate with chained Fusee Payload
- Latest Regular Hekate
- Direct fusee payload
(Tried all, resulted in same crash)

- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: None

SciresM, 2021-01-14:
Set nogc = 0 in BCT.ini.

closed 2021-01-14 by SciresM

TheRadziu, 2021-01-14:
It didnt fix the issue, just changed the program id where it crashes. 
Weirdly enough, the fix was to eject gamecard.

SciresM, 2021-01-14:
Oh, sure.

If you have ever played a gamecard on >= 11.0.0, you can never ever use gamecards on < 11.0.0. This is a hardware limitation.

TheRadziu, 2021-01-14:
sorry to reply so much but the issue was quite new to me, I was aware of this limitation but last time I used atmosphere (few months ago) it only couldnt run the gamecart but i did not crash whole system.

SciresM, 2021-01-14:
When booting emummc on firmware < 11.0.0, you probably want nogc = 1.

This will prevent system crash on gamecard inserted.

nbalonso, 2021-01-22:
This caught me too. Have been running the latest fw and playing cartridges normally for months and after a long time booted emummc and was crashing.

Booting with no cartridge and upgrading the fw fixed the problem but I wish the fatal screen had told me what the problem was in a more human readable way

Is there a plan for simpler messages?