Atmosphere-NX/Atmosphere #1335: Reboot to payload doesn't work on Erista modchip, but works on Mariko modchip

MrVtR, 2021-01-14:
## Bug Report

Reboot option on Nintendo main menu doesn't work on Erista model to redirect to payload

### What's the issue you encountered?

When I try to reboot my Erista switch with modchip, it goes to an black screen and I'm forced to hold power until the switch shutdowns(10-15 seconds). With the same files, it worked on Mariko model (Switch Lite) and it could reboot to payload
This issue was found on Erista emuMMC, eMMC is working without any issues

### How can the issue be reproduced?

The issue can be reproduced by trying to reboot an emuMMC from Erista model with modchip installed 

### Crash Report

N/A

### System Firmware Version

11.0.1

### Environment?

Atmosphere 0.17.0/0.17.1(both with the same issue) and Hekate v5.5.2

### Additional context?

The same issue was related by one other user on the modchip Support Community on Discord

EmreTech, 2021-01-14:
I thought reboot to payload wouldn't work on mod chipped Eristas and Marikos.

MrVtR, 2021-01-14:
> I thought reboot to payload wouldn't work on mod chipped Eristas and Marikos.

Here just the .nro doesn't work on Mariko, but if I try to reboot with nintendo menu option, it goes to hekate after rebooting

SciresM, 2021-01-14:
Are you using spacecraft? What's your reboot payload.bin?

asbeserra, 2021-01-14:
I have the same problem

SciresM, 2021-01-14:
If you have the same problem: are you using spacecraft? What's your reboot payload.bin?

MrVtR, 2021-01-14:
> Are you using spacecraft? What's your reboot payload.bin?

I'm using sxgear as boot.dat with Hekate as payload.bin

SciresM, 2021-01-14:
not payload.bin, reboot_payload.bin inside atmosphere folder. What do you have as that?

I'm betting you're rebooting-to-fusee, which doesn't work without cleanup on non spacecraft, which isn't supported anyway.

MrVtR, 2021-01-14:
> not payload.bin, reboot_payload.bin inside atmosphere folder. What do you have as that?
> 
> I'm betting you're rebooting-to-fusee, which doesn't work without cleanup on non spacecraft, which isn't supported anyway.

I'm using the original reboot_payload that comes with atm 0.17, [this link here on atm repo](https://github.com/Atmosphere-NX/Atmosphere/releases/download/0.17.0/atmosphere-0.17.0-master-e53b9cc5+hbl-2.4.0+hbmenu-3.4.0.zip)

SciresM, 2021-01-14:
reboot_payload default is fusee, which would have issue as described.

So, yeah, switch to spacecraft or use a different reboot payload.

closed 2021-01-14 by SciresM

asbeserra, 2021-01-14:
what would be the payload to be used? I'm also using hakete, gear and I'm not getting it, to make it worse I tried to use spacecraft and I couldn't, some incompatibility of the zadig drive with windows. forgive my terrible english, i'm brazilian (v1 erista modchip)