Atmosphere-NX/Atmosphere #1337: gamecard reader doesn

pcwizard7, 2021-01-15:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2021-01-15:
set nogc = 0 in BCT.ini, or edit hekate configuration to set nogc = 0 if booting from hekate.

closed 2021-01-15 by SciresM

pcwizard7, 2021-01-15:
Still broken.

Also works fine in stock. but boot into ATMOS Sysnand read errors

SciresM, 2021-01-15:
I guarantee you this is a failure to properly edit configuration.

Show screenshot of your atmosphere/config folder and paste the contents of your bct.ini?

If booting via hekate, paste your hekate_ipl.ini?

pcwizard7, 2021-01-15:
atmosphere/config folder is empty 

Hekate_ipl.ini
> [config]
autoboot=0
autoboot_list=0
bootwait=2
backlight=54
autohosoff=1
autonogc=0
updater2p=0
bootprotect=0

[ATMOSPHERE (sysNAND)]
payload=bootloader/payloads/fusee-primary.bin
icon=bootloader/res/atmosphere.bmp

[ATMOSPHERE (emuNAND)]
payload=bootloader/payloads/fusee-primary.bin
icon=bootloader/res/atmosphere-emu.bmp

[SXOS CFW]
payload=bootloader/payloads/SX_Loader.bin
icon=bootloader/res/sxos.bmp

[Stock]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
icon=bootloader/res/icon_switch.bmp
stock=1

SciresM, 2021-01-15:
"atmosphere/config folder is empty"

so, as expected, you have not edited BCT.ini.

copy the template into atmosphere/config and add nogc = 0 in the right place.

pcwizard7, 2021-01-15:
thanks, I didn't know it needed to be in there since it was in Hekate