Atmosphere-NX/Atmosphere #1338: Super Mario Odyssey Action Guide softlock console

cybercraft-fr, 2021-01-16:
## Bug Report

### What's the issue you encountered?

Go to the Action Guide in Super Mario Odyssey softlock the entire console, forcing a hard reboot.

### How can the issue be reproduced?

From menu
Select 'Action Guide' from the main menu and quit with B button

### Crash Report

"N/A"

### System Firmware Version

11.0.1

### Environment?

- bootloader retroreloaded:
  -  version Atmosphere 0.16.2-master-1085ccbe
   - retroreloaded's distribution of Atmosphère
- Sysmodules disabled: sys-con, sys-ftpd-light, Tesla, ldn_mitm

### Additional context?

The symptoms of this issue seem similar to #775 but I have same problem with sys-con disabled.

SciresM, 2021-01-16:
"Disabled" isn't a real thing.

does the issue occur using ams release zip with no custom sysmodules in /contents/?

kbmarinha, 2021-01-18:
I couldnt reproduce this issue. Its working fine here.

SciresM, 2021-01-19:
Closing due to issue opener non-responsiveness, inability to reproduce, and likelihood that this is a custom sysmodule issue.

closed 2021-01-19 by SciresM